TheThingsNetwork/lorawan-stack #301: permission deniedlorawan-stack_300-rc2_linux_amd64_stack_1 exited with code 255

ramesh3443, 2019-03-18:
Dear Developer, 

When i run the doc composer i am having some errors can you please check it.

Starting lorawan-stack_300-rc2_linux_amd64_cockroach_1 ... done
Starting lorawan-stack_300-rc2_linux_amd64_redis_1     ... done
Starting lorawan-stack_300-rc2_linux_amd64_stack_1     ... done
Attaching to lorawan-stack_300-rc2_linux_amd64_cockroach_1, lorawan-stack_300-rc2_linux_amd64_redis_1, lorawan-stack_300-rc2_linux_amd64_stack_1
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | *
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | *
cockroach_1  | * Check out how to secure your cluster: https://www.cockroachlabs.com/docs/v2.1/secure-a-cluster.html
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "94bcb2928c79" to other nodes, is this routable?
cockroach_1  | *
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | *
cockroach_1  | *
redis_1      | 1:C 18 Mar 2019 15:37:30.909 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 18 Mar 2019 15:37:30.909 # Redis version=5.0.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 18 Mar 2019 15:37:30.909 # Configuration loaded
redis_1      | 1:M 18 Mar 2019 15:37:30.912 * Running mode=standalone, port=6379.
cockroach_1  | CockroachDB node starting at 2019-03-18 15:37:30.97352968 +0000 UTC (took 0.7s)
cockroach_1  | build:               CCL v2.1.1 @ 2018/11/19 18:24:21 (go1.10.3)
cockroach_1  | webui:               http://94bcb2928c79:26256
cockroach_1  | sql:                 postgresql://root@94bcb2928c79:26257?sslmode=disable
cockroach_1  | client flags:        /cockroach/cockroach <client cmd> --host=94bcb2928c79:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp933122357
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | status:              restarted pre-existing node
cockroach_1  | clusterID:           5aaf5409-1ca8-4a82-89fe-8e2e6e39127f
cockroach_1  | nodeID:              1
redis_1      | 1:M 18 Mar 2019 15:37:30.912 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1      | 1:M 18 Mar 2019 15:37:30.912 # Server initialized
redis_1      | 1:M 18 Mar 2019 15:37:30.912 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
redis_1      | 1:M 18 Mar 2019 15:37:30.912 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1      | 1:M 18 Mar 2019 15:37:30.912 * Ready to accept connections
stack_1      |   INFO Setting up core component
stack_1      |   WARN No cookie hash key configured, generated a random one hash_key=2A12AA6DC5AAD5A5FBE044618505A993AAE8F92774C2C84D610D4AACDA52461D91AB5B2429FB2FC5761EAB8A5C0A9454446A99048A4F03941D911A9D46A2B92B namespace=web
stack_1      |   WARN No cookie block key configured, generated a random one block_key=1D9ADAC26458DB604488B65B5EBD8C1A2857B8E66D6A57D0D8A9BDA2D686EAD2 namespace=web
stack_1      |   INFO Setting up Identity Server
stack_1      |   INFO Setting up Gateway Server
stack_1      | error:cmd/internal/shared:initialize_gateway_server (could not initialize Gateway Server)
stack_1      | --- error:pkg/gatewayserver:listen_frontend (failed to start frontend listener `tls` on address `:8882`)    protocol=tls
stack_1      |     address=:8882
stack_1      | --- open /run/secrets/key.pem: permission deniedlorawan-stack_300-rc2_linux_amd64_stack_1 exited with code 255

Regards
Ramesh

johanstokking, 2019-03-18:
@ramesh3443 please follow the issue template and add steps to reproduce.

You either didn't generate the certificates, or the file permissions are not right. See [Getting Started](http://github.com/thethingsnetwork/lorawan-stack/tree/master/doc/gettingstarted.md).

closed 2019-03-18 by johanstokking

ramesh3443, 2019-03-18:
Hi Johan,

Please see the attached screenshot.
I have generated the certificates
[image]
.

johanstokking, 2019-03-18:
What are the file permissions? `$ ls -l`

johanstokking, 2019-03-18:
Can you please update your original post with the issue template? Then we can reopen this and go ahead with support.