TheThingsNetwork/lorawan-stack #304: open /run/secrets/key.pem: permission deniedlorawan-stack_300-rc2_linux_amd64_stack_1 exited with code 255

ramesh3443, 2019-03-18:
**Summary:** 

permission deniedlorawan-stack_300-rc2_linux_amd64_stack_1 exited with code 255
...

**Steps to Reproduce:** 

1. ... I have updated the server with .cert.pem and key.pem
2. ...Followed the docker commands
3. ...when i run the command docker-compose up, i am getting the following error
redis_1      | 1:C 18 Mar 2019 17:21:21.846 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 18 Mar 2019 17:21:21.846 # Redis version=5.0.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 18 Mar 2019 17:21:21.846 # Configuration loaded
redis_1      | 1:M 18 Mar 2019 17:21:21.847 * Running mode=standalone, port=6379.
redis_1      | 1:M 18 Mar 2019 17:21:21.847 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1      | 1:M 18 Mar 2019 17:21:21.847 # Server initialized
redis_1      | 1:M 18 Mar 2019 17:21:21.847 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
redis_1      | 1:M 18 Mar 2019 17:21:21.847 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1      | 1:M 18 Mar 2019 17:21:21.847 * Ready to accept connections
stack_1      |   INFO Setting up core component
stack_1      |   WARN No cookie hash key configured, generated a random one hash_key=EBDA1370613428BCE4D192CA03C56568409B9E95A9A7E32C9B401D6C6E00388E0748170426AF883A346A5ACE87CEAA6744808DD7403721CD0EDB38F737DB6ECE namespace=web
stack_1      |   WARN No cookie block key configured, generated a random one block_key=5AE8BB77CB360A1CD06916DD8727EFD66C8853C3745B19E63294CA7E028A9C6C namespace=web
stack_1      |   INFO Setting up Identity Server
stack_1      |   INFO Setting up Gateway Server
stack_1      | error:cmd/internal/shared:initialize_gateway_server (could not initialize Gateway Server)
stack_1      | --- error:pkg/gatewayserver:listen_frontend (failed to start frontend listener `tls` on address `:8882`)    address=:8882
stack_1      |     protocol=tls
stack_1      | --- open /run/secrets/key.pem: permission deniedlorawan-stack_300-rc2_linux_amd64_stack_1 exited with code 255

**What do you see now?**

...
[image]
[image]

**What do you want to see instead?**

I have to see the Lorawan stack should run
...

**How do you propose to implement this?**

...

**Environment:**

OS  UBUNTU 18.04
Linux ip-172-31-4-33 4.15.0-1032-aws #34-Ubuntu SMP Thu Jan 17 15:18:09 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

...

**What can you do yourself and what do you need help with?**

...

johanstokking, 2019-03-18:
@ramesh3443 thanks for filling out the issue template. I'm afraid we have to be strict with this to avoid precedence and it gives us the required information.

From what I see, the certificate and key are owned by `root` and your Docker container doesn't have access. You can try changing ownership (see `man chown`) of the files to `ubuntu`.

Did you follow these steps also? https://docs.docker.com/install/linux/linux-postinstall/

In any case, this doesn't seem like a TTN specific issue.

closed 2019-04-04 by johanstokking

anthonybudd, 2022-11-20:
If anyone else is having this issue on Mac, I discovered that when you follow the [instructions](https://www.thethingsindustries.com/docs/getting-started/installation/certificates/), `key.pem` file cannot be opened without `sudo`.

This will produce this error message in docker:
```sh
network-stack-1     | ERROR	Could not start gRPC server	{"error": "error:pkg/component:listener (could not create `gRPC/tls` listener)", "error_cause": "open /run/secrets/key.pem: operation not permitted", "protocol": "gRPC/tls"}
...
network-stack-1     | error:pkg/component:listener (could not create `gRPC/tls` listener)
network-stack-1     |     protocol=gRPC/tls
network-stack-1     |     correlation_id=c3de686af5c8449886d9e427c31e2be0
network-stack-1     | --- open /run/secrets/key.pem: operation not permitted
network-stack-1 exited with code 255
```

Simply running `sudo chown 886:886 ./cert.pem ./key.pem` will not fix this.

I know there's probably a better way to do this but as a quick fix I did the following:
1. `sudo cat key.pem`
2. **copy the body of the key file.
3. `mv key.pem key_.pem`
4. `nano key.pem`
5. **paste
6. **save and exit nano
7. `sudo chown 886:886 ./key.pem`
8. `docker compose up`