TheThingsNetwork/lorawan-stack #313: NS warns that session is empty after OTAA join

johanstokking, 2019-03-19:
**Summary:** 

NS warns that session is empty after OTAA join.

**Steps to Reproduce:** 

1. Checkout current tip `03696ccb`
2. Start stack; `ttn-lw-stack start`
3. Create device; `ttn-lw-cli dev create test-app simulation --lorawan-version 1.0.2 --lorawan-phy-version 1.0.2-b --frequency-plan-id EU_863_870 --join-eui 4200000000000000 --dev-eui 4200000000000000 --root-keys.app-key.key 42000000000000000000000000000000 --supports-class-c`
4. Run the simulator (see https://github.com/johanstokking/mbed-os-lorawan-simulation)
5. Schedule downlink; `mosquitto_pub -h localhost -u test-app -P $APP_API_KEY -d -t 'v3/test-app/devices/simulation/down/push' -m '{"downlinks":[{"f_port":15,"frm_payload":"vu8="}]}'

**What do you see now?**

NS warns;

```
  WARN Failed to update device in registry      delay=7.289ms device_class=CLASS_C device_uid=test-app.simulation error=error:pkg/networkserver:empty_session (session in empty) start_at=2019-03-19 10:43:49.328041 +0000 UTC
  WARN Task failed                              error=error:pkg/networkserver:empty_session (session in empty) invocation=2 task_id=process_downlink
```

And nothing happens.

Note that the `session` is in fact empty;

```
$ ttn-lw-cli dev get test-app simulation3 --session.dev-addr --supports-class-c
{
  "ids": {
    "device_id": "simulation3",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "4200000000000002",
    "join_eui": "4200000000000000"
  },
  "created_at": "2019-03-19T10:45:00.106Z",
  "updated_at": "2019-03-19T10:45:00.106Z",
  "network_server_address": "localhost:1884",
  "application_server_address": "localhost:1884",
  "supports_class_c": true,
  "session": {
    "dev_addr": "00000000",
    "keys": {

    },
    "started_at": "0001-01-01T00:00:00Z"
  }
}
```

**What do you want to see instead?**

Having a session there and working downlink.

@adriansmares confirms the issue by reproduction of the steps above.

**How do you propose to implement this?**

Not sure. The session may not be written back to the registry. No errors or warnings are reported about this though.

**What can you do yourself and what do you need help with?**

Review.

rvolosatovs, 2019-03-19:
Session is empty, since the device has not confirmed the session switch by sending an uplink using the new session context.
I added support for multicast in #318

johanstokking, 2019-03-19:
This is not multicast; this is after a successful join. See the link in 4.

rvolosatovs, 2019-03-19:
Yes, I used wrong terminology - I meant downlink in an uplink-less OTAA session here(so without session confirmation)

johanstokking, 2019-03-28:
Turns out that this behavior is partly what we want. In fact, if `pending_session` is set, we need to refuse class C downlink.

PR #411 (2019-04-01): Set device class to B/C on session confirmation

**Summary:**

Closes #313 
Closes #404 

**Changes:**

- Do not set class B/C directly on join-accept; the device has to confirm the pending session via an uplink
- On uplink, set the desired device class
closed 2019-04-01 by johanstokking