TheThingsNetwork/lorawan-stack #315: Include created_at and updated_at when setting entities

johanstokking, 2019-03-19:
**Summary:** 

Include `created_at` and `updated_at` when setting entities.

**Why do we need this?**

To make behavior consistent when getting entities.

**What is already there? What do you see now?**

When setting a device, for example:

```
$ ttn-lw-cli dev set test-app simulation3 --supports-class-c=false
{
  "ids": {
    "device_id": "simulation3",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "4200000000000002",
    "join_eui": "4200000000000000"
  },
  "created_at": "0001-01-01T00:00:00Z",
  "updated_at": "0001-01-01T00:00:00Z"
}
```

These values are blanked out.

**What is missing? What do you want to see?**

The `created_at` and `updated_at` fields filled out.

**How do you propose to implement this?**

Add the fields, like `ids`.

**Environment:**

Any.

**What can you do yourself and what do you need help with?**

Review.

htdvisser, 2019-03-20:
What value should the CLI display in the response? Does it make sense to take the oldest `CreatedAt` and newest `UpdatedAt`?

johanstokking, 2019-03-20:
I guess this is very intuitive right?

- on create, `created_at` and `updated_at` are the same server time
- on update, `created_at` doesn't change and is the original value, and `updated_at` is the new server time

htdvisser, 2019-03-20:
My point was that with `EndDevice`s, we have a `created_at` and `updated_at` in all of the registries, and we can only display one of those values in the CLI response, because it merges the responses from all registries into a single `EndDevice` struct.

johanstokking, 2019-03-20:
Ah, I understand, yeah good suggestion.

htdvisser, 2019-03-21:
Created issue for that: https://github.com/TheThingsNetwork/lorawan-stack/issues/338

PR #344 (2019-04-03): Implicit Redis registry fieldmasks
**Summary:**

Closes #315 

**Changes:**

- Handle implicit fieldmasks in a saner way
- Only validate the entities actually stored in the registries, instead of performing validation every time we apply the fieldmask
- Actually perform validation in AS registries

**Notes for Reviewers:**

@johanstokking 	please fix the invalid AS tests
I suppose this is not going to be finished before my holiday, so assigning you

```
go run ./cmd/ttn-lw-cli login && go run ./cmd/ttn-lw-cli end-devices create --application-id test-app  --device-id test-sim --dev-eui 42000000 [... cut]
closed 2019-04-03 by johanstokking