TheThingsNetwork/lorawan-stack #316: Add sorting functionality to list requests

kschiffer, 2019-03-19:
**Summary:** 
RPC call returning a list should support a sort option, to sort by a specific field either ascending or descending.

**Why do we need this?**
Sorting only by the id is not always desirable. Console users will likely also want to sort by creation date or something else.

**What is already there? What do you see now?**
The list protos already have an interface for this:
```
  // Order the results by this field path (must be present in the field mask).
  // Default ordering is by ID. Prepend with a minus (-) to reverse the order.
  string order = 3;
  // Limit the number of results per page.
  uint32 limit = 4;
  // Page number for pagination. 0 is interpreted as 1.
  uint32 page = 5;
```

**What is missing? What do you want to see?**
However, setting the sort value does not appear to have any effect yet. (While limit and page values work as expected)

**How do you propose to implement this?**
@htdvisser 

**Environment:**
Making requests via the http bridge of gRPC.

**What can you do yourself and what do you need help with?**
Help.

htdvisser, 2019-03-20:
Implementation wise this would involve translating the field mask path (which is the value of the `order` field) to the database column name and passing that to the SQL `ORDER` instead of `ids`.

johanstokking, 2019-06-20:
Pushing to Backlog. Let's consider this again based on user feedback

rish-c, 2019-12-11:
Sorting gateways would be good to add in the console. Possible fields could be
- Connected / Not connected
- Data added
- Owner
- labels (location, site, indoor, outdoor etc.)

cc @kschiffer

htdvisser, 2019-12-12:
It's relatively easy to implement sorting on:

- ID
- EUIs (Gateways, End Devices)
- Name
- Created At

It's more work to implement sorting on:

- Attributes

It's currently **not** possible to sort on:

- Owner: we don't have the concept of owner in v3. We have (multiple) members with (different) rights and we can't sort on those.
- Connected (Gateways): This is not known by the Entity Registry.
- Last Seen (Gateways, End Devices): This is not known by the Entity Registry.

kschiffer, 2019-12-12:
Frontend-wise it's more or less a matter of re-activating our logic for sorting in our list views, so this is mostly blocked on the backend side.

PR #1785 (2019-12-27): Ordering in List RPCs

#### Summary

This Pull Requests resolves #316 by adding support for ordering in list RPCs.

#### Changes

- Add validation rules to `order` fields
- Propagate ordering from IS RPCs to stores
- Rewrite some IS store queries

#### Notes for Reviewers

Main reviewer is @bafonins. 

Strictly speaking we're breaking API because we're now returning errors on certain inputs for the `order` field that we allowed in the past (because there was no validation whatsoever). An alternative would be to silently fail on invalid values for the field, but I think that would be a bug whic [... cut]
closed 2019-12-27 by htdvisser