TheThingsNetwork/lorawan-stack #321: AS923 should have CFListType = 0

adamsondelacruz, 2019-03-20:
**Summary:** 

An error is encountered when a device configured for AS923 is using the V3 stack

**Steps to Reproduce:** 

1. Set the gateway configuration to use AS923
2. Set the device to use AS923
3. Register the gateway and device(OTAA) in a running v3 stack
4. Do a JOIN and you can see from the console log, errors

**What do you see now?**

...

**What do you want to see instead?**

...

**How do you propose to implement this?**

As suggessted by Hylke, the line pkg/encoding/lorawan/messages.go:242 fails because the message is falling under CFListType = 1 whereas AS923 should have CFListType = 0
...

**Environment:**

Mac OS, Chrome, mDot device running AS923 FW, Gateway running classic packet forwarder. 
...

**What can you do yourself and what do you need help with?**

...

rvolosatovs, 2019-03-20:
The issue was with invalid `band` definitions: see #324

ama9910, 2019-03-25:
No longer crashing, but a different error appears after this change. 

`INFO` Finished unary call                      duration=1.936659ms error=error:pkg/joinserver:encode_payload (failed to encode payload) error_cause=error:pkg/encoding/lorawan:encode (could not encode 'CFList') error_cause_cause=error:pkg/encoding/lorawan:field_length_with_max ('CFListFreq' length '6' should be lower or equal to '5') ...`

johanstokking, 2019-03-26:
@adriansmares the issue is that 8-2=6 frequencies are sent as `CFList`, while only 5 are supported. The sixth one should come as `NewChannelReq`.

PR #324 (2019-03-20): Fix invalid CFListType in bands
**Summary:**

Closes #321 

**Changes:**

- Fix invalid `CFListType` values in bands

**Notes for Reviewers:**

I went through all the bands in 1.1 rev B spec and made sure `CFListType` is consistent in `band` package with the spec
PR #438 (2019-04-04): util: Limit the CFList to 5 frequencies

**Summary:**

Closes #321.

**Changes:**

- Use at most 5 frequencies in the `JoinRequest` CFList

**Notes for Reviewers:**

None.

**Release Notes: _(optional)_**

None.
closed 2019-04-04 by johanstokking