TheThingsNetwork/lorawan-stack #327: Make JS aware of JoinEUI ranges

johanstokking, 2019-03-20:
**Summary:** 

Make the Join Server aware of its JoinEUI ranges and make this available to clients.

**Why do we need this?**

To validate the created devices and inform clients which JoinEUIs are supported.

**What is already there? What do you see now?**

No information about JoinEUIs for the JS.

**What is missing? What do you want to see?**

Reporting of;
- JoinEUI ranges (`[]types.EUI64Prefix`) of accepted join-requests
- JoinEUI ranges (`[]types.EUI64Prefix`) of accepted new devices

**How do you propose to implement this?**

Add configuration, extend API, implement RPC server.

**What can you do yourself and what do you need help with?**

Can do, can review.

johanstokking, 2019-08-05:
@rvolosatovs don't we have this already?

rvolosatovs, 2019-08-05:
Done in https://github.com/TheThingsNetwork/lorawan-stack/pull/803

closed 2019-08-05 by rvolosatovs