TheThingsNetwork/lorawan-stack #328: Fetch JoinEUIs from JS and use accepted default for device create

johanstokking, 2019-03-20:
**Summary:** 

Fetch the accepted JoinEUI ranges for new devices (#327) and use those for new devices.

**Why do we need this?**

For the Console to fill out the default JoinEUI for a new device, and support client side validation.

**What is already there? What do you see now?**

Nothing.

**What is missing? What do you want to see?**

Fetch the accepted ranges of JoinEUIs.

Note that you'll get an array of ranges; as a default JoinEUI, pick the lowest accepted in the first range.

**How do you propose to implement this?**

@kschiffer 

**What can you do yourself and what do you need help with?**

Test

johanstokking, 2019-03-20:
Blocked by #327

kschiffer, 2019-08-05:
Superseded by https://github.com/TheThingsNetwork/lorawan-stack/issues/820

closed 2019-08-05 by kschiffer