TheThingsNetwork/lorawan-stack #334: Add component logging namespaces

johanstokking, 2019-03-20:
**Summary:** 

Set component namespace to logging field in context.

**Why do we need this?**

To easily separate log entries by component.

**What is already there? What do you see now?**

In places where we control the context, we already set the namespace like `gatewayserver/io/udp`.

**What is missing? What do you want to see?**

RPC calls and other requests do not contain this as we have one RPC server and one web server.

**How do you propose to implement this?**

Add gRPC middleware to set the `namespace` log field on a derived context and pass that context to the next call.

We may have to stringtype to know which service is exposed by which component.

**What can you do yourself and what do you need help with?**

Review

**Discussion**

@htdvisser if you have better ideas, feel free to share.

johanstokking, 2019-03-20:
Another idea; when the components register their services, we may insert some indirection layer that does the actual registration plus record the service name and the calling component.

htdvisser, 2019-03-20:
That would be nice and IMO cleaner, but it's not possible with grpc's generated code, which really expects a `*grpc.Server` and not an interface.

```go
func RegisterXxxServer(s *grpc.Server, srv XxxServer) {
```

htdvisser, 2019-03-20:
I think we can do this with the existing `hooks` middleware. We can add the following to `pkg/rpcmiddlware/rpclog`:

```go
const NamespaceHook = "namespace"

func UnaryNamespaceHook(namespace string) hooks.UnaryHandlerMiddleware {
	return func(h grpc.UnaryHandler) grpc.UnaryHandler {
		return func(ctx context.Context, req interface{}) (interface{}, error) {
			ctx = log.NewContextWithField(ctx, "namespace", namespace)
			return h(ctx, req)
		}
	}
}

func StreamNamespaceHook(namespace string) hooks.StreamHandlerMiddleware {
	return func(h grpc.StreamHandler) grpc.StreamHandler {
		return func(srv interface{}, stream grpc.ServerStream) error {
			wrapped := grpc_middleware.WrapServerStream(stream)
			wrapped.WrappedContext = log.NewContextWithField(stream.Context(), "namespace", namespace)
			return h(srv, wrapped)
		}
	}
}
```

And in the components, we can call something like:

```go
hooks.RegisterUnaryHook("/ttn.lorawan.v3.Ns", log.NamespaceHook, log.UnaryNamespaceHook("ns"))
hooks.RegisterStreamHook("/ttn.lorawan.v3.Ns", log.NamespaceHook, log.StreamNamespaceHook("ns"))
```

PR #581 (2019-04-30): Component logging namespaces

**Summary:**

Closes #334

**Changes:**

- Added the component namespaces logging hooks to `rpcmiddleware/rpclog`
- Added said hooks to GS/NS/IS/JS

**Notes for Reviewers:**

I've implemented it based on @htdvisser suggestion, with a slight change in component names (`GsNs` instead of simply `Ns` and so forth).

**Release Notes:**

- Added logging namespaces to the gRPC calls.
closed 2019-04-30 by htdvisser