TheThingsNetwork/lorawan-stack #335: NS fails to handle MAC answers on class C downlink

johanstokking, 2019-03-20:
**Summary:** 

Network Server fails to handle MAC answers that are sent from the device on class C downlinks.

**Steps to Reproduce:** 

1. Checkout `8e422088126fe78774f6c93fbbd83be30e39ad33` (#318 + #333)
2. Start stack
3. Create device supporting class C; `ttn-lw-cli dev create test-app simulation --lorawan-version 1.0.2 --lorawan-phy-version 1.0.2-b --frequency-plan-id EU_863_870 --join-eui 4200000000000000 --dev-eui 4200000000000000 --root-keys.app-key.key 42000000000000000000000000000000 --supports-class-c`
4. Run the [simulation device](https://github.com/johanstokking/mbed-os-lorawan-simulation); i.e. OTAA join + switch to class C. Do not press the button
5. Send a downlink message; `ttn-lw-cli dev downlink push test-app simulation --f-port=15 --frm-payload=0101`

**What do you see now?**

An application downlink message is sent carrying MAC commands that require an answer. The device is answering, but this is not handled by the NS.

This is writing the class C downlink message, this is all good:

```
 DEBUG Pushed application downlink to queue     device_class=CLASS_C device_uid=test-app.simulation grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AsNs namespace=grpc queue_length=1 request_id=01D6E4A68XWY09W6S20BWWZXQ5
 DEBUG Adding downlink task...                  device_class=CLASS_C device_uid=test-app.simulation grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AsNs namespace=grpc queue_length=1 request_id=01D6E4A68XWY09W6S20BWWZXQ5
  INFO Finished unary call                      duration=26.82446ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AsNs namespace=grpc request_id=01D6E4A68XWY09W6S20BWWZXQ5
 DEBUG Finished unary call                      duration=27.429983ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AsNs namespace=grpc request_id=01D6E4A67JBHWTFM60DA02KCFF
  INFO Finished unary call                      duration=73.690216ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=grpc request_id=01D6E4A67JBHWTFM60DA02KCFF
 DEBUG Processing downlink task...              device_uid=test-app.simulation start_at=2019-03-20 18:02:47.991386 +0000 UTC started_at=2019-03-20 19:02:47.996774 +0100 CET m=+24.004955567
 DEBUG transport: loopyWriter.run returning. connection error: desc = "transport is closing" namespace=grpc
 DEBUG Generated downlink                       ack=false device_uid=test-app.simulation f_cnt=1 f_pending=true f_port=0 m_type=UNCONFIRMED_DOWN mac_count=7 payload_length=46 start_at=2019-03-20 18:02:47.991386 +0000 UTC started_at=2019-03-20 19:02:47.996774 +0100 CET m=+24.004955567
 DEBUG Scheduling downlink...                   attempt_rx1=false attempt_rx2=true device_class=CLASS_C device_uid=test-app.simulation downlink_type=data path_count=1 start_at=2019-03-20 18:02:47.991386 +0000 UTC started_at=2019-03-20 19:02:47.996774 +0100 CET m=+24.004955567
 DEBUG Attempting to schedule downlink on gateway class=CLASS_C gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp
 DEBUG Attempting to schedule downlink in receive window class=CLASS_C data_rate=0 frequency=0 gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp rx_window=1
 DEBUG Attempting to schedule downlink in receive window class=CLASS_C data_rate=0 frequency=869525000 gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp rx_window=2
 DEBUG Scheduled downlink                       class=CLASS_C data_rate=0 duration=2.301952s frequency=869525000 gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp now=18531466355 rx_window=2 starts=18861461000
 DEBUG Writing downlink message                 gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp remote_addr=127.0.0.1:58961
  INFO Finished unary call                      duration=2.106523ms grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc request_id=01D6E4A6A5DR3Y3QQG5Z9HF6PS
 DEBUG Finished unary call                      duration=2.801556ms grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc
 DEBUG Scheduled downlink                       attempt_rx1=false attempt_rx2=true delay=329.994645ms device_class=CLASS_C device_uid=test-app.simulation downlink_type=data start_at=2019-03-20 18:02:47.991386 +0000 UTC started_at=2019-03-20 19:02:47.996774 +0100 CET m=+24.004955567
 DEBUG Adding device to downlink schedule...    device_class=CLASS_C device_uid=test-app.simulation schedule_at=2019-03-20 18:03:03.005065 +0000 UTC start_at=2019-03-20 18:02:47.991386 +0000 UTC started_at=2019-03-20 19:02:47.996774 +0100 CET m=+24.004955567
 DEBUG Recalculating downlink queue             application_uid=test-app count=1 device_uid=test-app.simulation network_server=Johans-MacBook-Pro.local next_a_f_cnt_down=2
 DEBUG Run database query                       duration=4ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=db query=SELECT * FROM "api_keys"  WHERE ("api_keys"."api_key_id" = $1) ORDER BY "api_keys"."id" ASC LIMIT 1 request_id=01D6E4A6AG91CFNY63ZS1J1C3S rows=1 source=api_key_store.go:71 values=[ILZVXXQBP4QVCPYPXIYVC6CPKBSAUOX65BW2V7Y]
 DEBUG Run database query                       duration=3ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=db query=SELECT id as uuid, application_id as friendly_id FROM "applications"  WHERE (id in ($1)) request_id=01D6E4A6AG91CFNY63ZS1J1C3S rows=1 source=membership.go:90 values=[ffe318d4-5750-44ec-903e-2068c42abe21]
 DEBUG Replaced application downlink queue      device_class=CLASS_C device_uid=test-app.simulation grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=grpc queue_length=1 request_id=01D6E4A6AG91CFNY63ZS1J1C3S
 DEBUG Adding downlink task...                  device_class=CLASS_C device_uid=test-app.simulation grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=grpc queue_length=1 request_id=01D6E4A6AG91CFNY63ZS1J1C3S
  INFO Finished unary call                      duration=32.985227ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=grpc request_id=01D6E4A6AG91CFNY63ZS1J1C3S
 DEBUG Finished unary call                      duration=33.510874ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=grpc
```

So far so good. Note the `delay=329.994645ms`; this is GS's `ScheduleTimeLong` (300 ms network latency + 30 ms gateway queue delay).

Question though; why is there a downlink queue invalidation here?

Then, the device comes back with a MAC answer:

```
 DEBUG Synchronized gateway timestamp           gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp time=2019-03-20 19:02:29.47362 +0100 CET m=+5.481751174 timestamp=20664000
 DEBUG Synchronized server absolute time only   gateway_eui=ACDE480000001122 gateway_uid=simulator namespace=gatewayserver/io/udp server_time=2019-03-20 19:02:50.137426 +0100 CET m=+26.145557961 timestamp=20664000
 DEBUG Deduplicating uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6E4A8CW0DY2J4F218ZVY0QZ
 DEBUG Handling data uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6E4A8CW0DY2J4F218ZVY0QZ
 DEBUG Matching device...                       dev_addr=00135E69 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6E4A8CW0DY2J4F218ZVY0QZ
 DEBUG Performing MIC checks on devices with matching frame counters... dev_addr=00135E69 device_count=1 device_uid=test-app.simulation full_f_cnt_up=1 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs last_f_cnt_up=0 namespace=grpc payload_length=23 request_id=01D6E4A8CW0DY2J4F218ZVY0QZ uplink_f_cnt=1 uplink_f_cnt_up=1
 DEBUG Matched device                           dev_addr=00135E69 device_uid=test-app.simulation grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6E4A8CW0DY2J4F218ZVY0QZ
 DEBUG Waiting for collection window to be closed... grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6E4A8CW0DY2J4F218ZVY0QZ
 DEBUG Collection window closed, stopped deduplication grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6E4A8CW0DY2J4F218ZVY0QZ
[... cut ...]
[DBG ][LRAD]:  ][LMAC]: RX1 slot open, Freq = 868500000
[DBG ][LMAC]: RX2 slot open, Freq = 869525000
[ERR ][LSTK]: Retries exhausted for Class C device
Transmission Error - EventCode = 4
```

Here you clearly see that the device is expecting an acknowledgement.

**What do you want to see instead?**

Successful flow.

**How do you propose to implement this?**

I think the downlink task is hanging in there too long.

**What can you do yourself and what do you need help with?**

Review and test a fix.

rvolosatovs, 2019-03-21:
The queue invalidations are sent, since you're using a pre-LoRaWAN1.1 device and downlink on `FPort=0` is sent by NS.
It looks like a bug in either the GS or the device itself - the NS behaves as expected, no uplink carrying MAC responses is received from the device and, hence, NS continuously reschedules the same MAC requests(and, hence, sends the invalidations)

Note, that GS returns negative delay:
```
 DEBUG Scheduled downlink                       attempt_rx1=false attempt_rx2=true delay=-32m19.550671274s device_class=CLASS_C device_uid=test-app.test-sim downlink_type=data start_at=2019-03-21 14:26:25.0541747 +0000 UTC started_at=2019-03-21 14:26:25.202742669 +0000 UTC
```

This is the log I'm getting in stack trying to reproduce this(with https://github.com/TheThingsNetwork/lorawan-stack/pull/318/commits/8d0ae82f9ec4c62066c9ad6d8efe5683aaa9a9b7 applied):
```
 DEBUG Processing downlink task...              device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding MAC command to buffer...          cid=CID_NEW_CHANNEL device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding MAC command to buffer...          cid=CID_NEW_CHANNEL device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding MAC command to buffer...          cid=CID_NEW_CHANNEL device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding MAC command to buffer...          cid=CID_NEW_CHANNEL device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding MAC command to buffer...          cid=CID_NEW_CHANNEL device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding MAC command to buffer...          cid=CID_RX_TIMING_SETUP device_uid=test-app.test-sim start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Generated downlink                       ack=false device_uid=test-app.test-sim f_cnt=34 f_pending=true f_port=0 m_type=UNCONFIRMED_DOWN mac_count=6 payload_length=45 start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Scheduling downlink...                   attempt_rx1=false attempt_rx2=true device_class=CLASS_C device_uid=test-app.test-sim downlink_type=data path_count=1 start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Attempting to schedule downlink on gateway class=CLASS_C gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp
 DEBUG Attempting to schedule downlink in receive window class=CLASS_C data_rate=0 frequency=0 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp rx_window=1
 DEBUG Attempting to schedule downlink in receive window class=CLASS_C data_rate=0 frequency=869525000 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp rx_window=2
 DEBUG Scheduled downlink                       class=CLASS_C data_rate=0 duration=2.138112s frequency=869525000 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp now=1640222325835 rx_window=2 starts=10840560000
  INFO Finished unary call                      duration=1.778692ms grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc request_id=01D6G9ZN26JWPTHCK1QJAR4CWP
 DEBUG Writing downlink message                 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp remote_addr=127.0.0.1:39305
 DEBUG Finished unary call                      duration=3.394588ms grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc
 DEBUG Scheduled downlink                       attempt_rx1=false attempt_rx2=true delay=-27m9.381765835s device_class=CLASS_C device_uid=test-app.test-sim downlink_type=data start_at=2019-03-21 14:20:22.78929 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Adding downlink task...                  device_class=CLASS_C device_uid=test-app.test-sim start_at=2019-03-21 14:20:37.98135339 +0000 UTC started_at=2019-03-21 14:20:22.958801903 +0000 UTC
 DEBUG Recalculating downlink queue             application_uid=test-app count=5 device_uid=test-app.test-sim network_server=neon next_a_f_cnt_down=35
 DEBUG Run database query                       duration=1ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=db query=SELECT * FROM "api_keys"  WHERE ("api_keys"."api_key_id" = $1) ORDER BY "api_keys"."id" ASC LIMIT 1 request_id=01D6G9ZN32EJGSD8YN09MJWAGZ rows=1 source=api_key_store.go:71 values=[W7MXMUE3UASTZKWLVED3QSY2CGWOFHHZM4UZQ6A]
 DEBUG Run database query                       duration=1ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=db query=SELECT id as uuid, application_id as friendly_id FROM "applications"  WHERE (id in ($1)) request_id=01D6G9ZN32EJGSD8YN09MJWAGZ rows=1 source=membership.go:90 values=[ae581d95-786e-40dd-b4b2-726fbdd733ab]
[... cut ...]
  INFO Finished unary call                      duration=27.839482ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=grpc request_id=01D6GA03X9Q4M4C49MZVMW52M9
 DEBUG Finished unary call                      duration=28.302477ms grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=grpc
 DEBUG Processing downlink task...              device_uid=test-app.test-sim start_at=2019-03-21 14:20:53.15796 +0000 UTC started_at=2019-03-21 14:20:53.227031549 +0000 UTC
...<repeated>...
```

rvolosatovs, 2019-03-21:
Here's what I see on 4fee6085ac642afc32ca3efcbf98780c8fe19c7d (`debug/downlink` branch):
```
 DEBUG Deduplicating uplink...                  grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Handling join-request...                 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Sending join-request to Join Server...   dev_addr=01D2A8F5 dev_eui=4200000000000000 device_uid=test-app.test-sim grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs join_eui=4200000000000000 namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
  INFO Finished unary call                      duration=2.308199ms grpc_method=HandleJoin grpc_service=ttn.lorawan.v3.NsJs namespace=grpc request_id=01D6H6R3F6DQR9KBM1GRDNQWSN
 DEBUG Finished unary call                      duration=2.804042ms grpc_method=HandleJoin grpc_service=ttn.lorawan.v3.NsJs namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Join-accept received from Join Server    dev_addr=01D2A8F5 dev_eui=4200000000000000 device_uid=test-app.test-sim grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs join_eui=4200000000000000 namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Sending join-accept to AS...             application_uid=test-app dev_addr=01D2A8F5 dev_eui=4200000000000000 device_uid=test-app.test-sim grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs join_eui=4200000000000000 namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Received AppSKey from Network Server     application_uid=test-app dev_eui=4200000000000000 device_uid=test-app.test-sim join_eui=4200000000000000 network_server=neon session_key_id=0169A26C0DE7B7563E5689C20FDFDC9B
 DEBUG Adding downlink task...                  application_uid=test-app dev_addr=01D2A8F5 dev_eui=4200000000000000 device_uid=test-app.test-sim grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs join_eui=4200000000000000 namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR start_at=2019-03-21 22:43:04.509290699 +0000 UTC
 DEBUG Waiting for collection window to be closed... grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Processing downlink task...              device_uid=test-app.test-sim start_at=2019-03-21 22:43:04.5092908 +0000 UTC started_at=2019-03-21 22:43:04.518866268 +0000 UTC
 DEBUG Scheduling downlink...                   attempt_rx1=true attempt_rx2=true device_class=CLASS_C device_uid=test-app.test-sim downlink_class=CLASS_A downlink_type=join-accept path_count=1 rx1_delay=5 rx1_frequency=868100000 rx2_data_rate=0 rx2_frequency=869525000 started_at=2019-03-21 22:43:04.518866268 +0000 UTC
 DEBUG Attempting to schedule downlink on gateway class=CLASS_A gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp
 DEBUG Attempting to schedule downlink in receive window class=CLASS_A data_rate=5 frequency=868100000 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp rx_window=1
 DEBUG Scheduled downlink                       class=CLASS_A data_rate=5 duration=71.936ms frequency=868100000 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp now=643028792021 rx_window=1 starts=647795000000
  INFO Finished unary call                      duration=1.831558ms grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc request_id=01D6H6R3PBNVQXG8P67FJZ0G4F
 DEBUG Writing downlink message                 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp remote_addr=127.0.0.1:34525
 DEBUG Finished unary call                      duration=3.496896ms grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc
 DEBUG Scheduled downlink                       attempt_rx1=true attempt_rx2=true delay=4.766207979s device_class=CLASS_C device_uid=test-app.test-sim downlink_class=CLASS_A downlink_type=join-accept rx1_delay=5 rx1_frequency=868100000 rx2_data_rate=0 rx2_frequency=869525000 started_at=2019-03-21 22:43:04.518866268 +0000 UTC
 DEBUG Received Tx acknowledgement, JIT queue supported gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp

 DEBUG Collection window closed, stopped deduplication grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
  INFO Finished unary call                      duration=1.201010048s grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01D6H6R3F46ZR7SZQ0NQAC3AVR
 DEBUG Finished unary call                      duration=1.20238852s grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc

 DEBUG Run database query                       duration=1ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=db query=SELECT * FROM "access_tokens"  WHERE ("access_tokens"."token_id" = $1) ORDER BY "access_tokens"."id" ASC LIMIT 1 request_id=01D6H6RGJVSQ4RG455JK6T0HQW rows=1 source=oauth_store.go:207 values=[HW2IBJWDJDHLUVWNANQUKS5VPB3456JL3CF6LUI]
 DEBUG Run database query                       duration=1ms grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=db query=SELECT * FROM "clients"  WHERE "clients"."deleted_at" IS NULL AND (("id" IN ($1))) ORDER BY "clients"."id" ASC request_id=0 [... cut ...]
 DEBUG Scheduled downlink                       class=CLASS_A data_rate=5 duration=46.336ms frequency=867100000 gateway_eui=0000000000000000 gateway_uid=eui-0000000000000000 namespace=gatewayserver/io/udp now=658724377804 rx_window=1 starts=663500000000
```
makes me believe that NS uses correct timing, since the `starts` is after `now`.
Also, GS still has timing issues (https://github.com/TheThingsNetwork/lorawan-stack/pull/333#issuecomment-475429490), so overall it seems like timing issue on GS side of things.

rvolosatovs, 2019-03-21:
Also, please note the `delay=1h11m26.860490949s` in the last log in the top-level post. (`DEBUG Scheduled downlink                       attempt_rx1=true attempt_rx2=false delay=1h11m26.860490949s device_class=CLASS_C device_uid=test-app.simulation downlink_class=CLASS_A downlink_type=data rx1_delay=5 rx1_frequency=868500000 start_at=2019-03-20 18:03:03.005065 +0000 UTC started_at=2019-03-20 19:03:03.241296 +0100 CET m=+39.249121497`)
This is clearly too much

rvolosatovs, 2019-03-21:
Blocked by #306

PR #379 (2019-03-29): Fix GS scheduling
**Summary:**

Replaces #333 
Closes #306 
Closes #335 
Closes #362

**Changes:**

- Fix synchronizing with the gateway timestamp
- Improve logging for debugging
closed 2019-03-29 by johanstokking