TheThingsNetwork/lorawan-stack #342: Remove generated message populators from pkg/ttnpb

htdvisser, 2019-03-21:
**Summary:** 

The generated message populators in `pkg/ttnpb` cause more problems than they solve. In my experience they're only causing flaky tests because of failed validations and non-deterministic code coverage because of randomly taken branches.

**What do you see now?**

- A mix of generated populators (that usually build garbage) and handwritten populators (that to a slightly better job)
- Randomly failing (or randomly passing) Travis builds
- Changes in code coverage without any code changes

**What do you want to see instead?**

The end goal is that our tests become deterministic, so ideally I'd want to see (handwritten or rule-aware) populators that have correct and deterministic output for a PRNG with the same seed.

**How do you propose to implement this?**

- Investigate if we can get more deterministic results with a PRNG with a static seed.
- Get rid of generated populators that build invalid structs
- Investigate if we can generate populators with the validation rules as constraints

**What can you do yourself and what do you need help with?**

I think it's best if @rvolosatovs looked into this, because `protoc` and `ttnpb` is his domain. Let me know if you need any help.

rvolosatovs, 2019-03-21:
Populators are fully deterministic.
The issue is that in some tests we run those concurrently, which is where the non-determinism comes from.

nejraselimovic, 2021-06-07:
@htdvisser is this important enough to leave it scheduled for Q2?

htdvisser, 2021-06-08:
We're slowly starting the big refactoring of `pkg/ttnpb`. The end goal is to migrate away from `gogo/protobuf` completely, and removing the message populators (that are generated by `gogo/protobuf`) is one part of that. 

I don't really enjoy doing these kind of chores, so I'd rather just get it over with than keep pushing it forward.

htdvisser, 2021-07-26:
Pull request #4450 removed all unused populators, so now it's time to remove all usage of the remaining ones, and remove those too. From what I can see, we still use populators in these places:

- `pkg/applicationserver/io/pubsub` - @adriansmares 
- `pkg/encoding/lorawan` -- @johanstokking
- `pkg/gatewayserver/io/udp` -- @KrishnaIyer 
- `pkg/identityserver/store` -- @htdvisser 
- `pkg/joinserver` -- @johanstokking 
- `pkg/networkserver` -- @adriansmares 
- `pkg/networkserver/mac` -- @adriansmares

johanstokking, 2021-07-27:
@KrishnaIyer for removing populators from `pkg/gatewayserver/io/udp`; just remove `generatePushData()` and instantiate uplink messages as needed.

@adriansmares the stuff in `pkg/networkservers` seems really easy to clean up; the populators are just there to have a valid state, not to do anything with it.