TheThingsNetwork/lorawan-stack #356: Support uplink retransmissions

rvolosatovs, 2019-03-23:
**Summary:** 

NS uplink matching does not support retransmissions of uplinks

**Why do we need this?**

To be fully compliant with LoRaWAN and make users happy

**What is already there? What do you see now?**

No retransmission support

**What is missing? What do you want to see?**

Retransmission support

**How do you propose to implement this?**

Use `NbTrans` parameter and a counter to count the number of (possible) retransmissions and drop uplinks exceeding the `NbTrans`.
If retransmission is encountered, the device MAC state should probably be rolled back, since it implies that the downlink(if any) was not received.
We may want to add additional "retransmission" message to send to AS from NS.

johanstokking, 2019-03-26:
Duplicate of #325 ?

rvolosatovs, 2019-04-15:
No, #325 was about a confusing logging statement. This one is about supporting the feature.

PR #653 (2019-05-10): Support uplink retransmissions

#### Summary

Closes #356 

#### Changes

- Add support for uplink retransmissions

#### Notes for Reviewers

More thorough testing will be done as part of #176 

#### Release Notes

- Added support for uplink retransmissions
closed 2019-05-10 by rvolosatovs