TheThingsNetwork/lorawan-stack #357: AS Downlink Fnct counter used on LoRaWAN 1.0.x devices resulting in dropped downlinks

pdubois22, 2019-03-25:
**Summary:** 
On 1.0.x LoRaWAN devices, AS downlink are transmitted with the AS downlink fnct which is incremented only on each AS downlink. 
On parallel, downlinks on port 0 use the NS downlink counter which is incremented on each NS downlink (typically MAC on port 0)
This behavior is per LoRaWan 1.1 spec, but is broken on a LoRaWAN 1.0 device because the ns counter will end up > to the as counter pretty quickly. Therefore, since there is only one counter on the device, the next AS downlink will be dropped at the device

This is a pretty serious pb for LoRaWAN 1.0.x devices

**Steps to Reproduce:** 
create a LoRaWAN 1.0 device
send a downlink
the downlink FnCt is the AS downlink which may be < to the NS downlink which has been received by the device.

**What do you see now?**
too small fnct and dropped downlinks on the device

**What do you want to see instead?**
one downlink fnct for 1.0 devices

**How do you propose to implement this?**
good question

**Environment:**
any

**What can you do yourself and what do you need help with?**
report it - done
let the masters of this code figure out the implications of AS Queue invalidation and crypted payload in the AS based on Fnct for LoRaWAN 1.0.x devices

johanstokking, 2019-03-25:
@pdubois22 thanks for reporting.

There are two separate counters, indeed because of compatibility with 1.1. However, we have accommodated for this through downlink queue invalidation, which happens when NS sends downlink with `FPort` of `0`. So it should work in fact.

Can you provide steps to reproduce per the issue template?

pdubois22, 2019-03-26:
@johanstokking please find the reproduction scenario - let me know if you need more

Step 1: Register an OTAA device LoRaWAN 1.0
===================

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices create caf 850706c7689521a801689b5f7a1b4ec5 --defaults=true \
--supports-join=true \
--join-eui=70B3D59520000000 --dev-eui=A81758FFFE0367C0 \
--root-keys.app-key.key=367A435DD86DF46DDA51AE51380872A8 \
--lorawan-version=1.0.2 --lorawan-phy-version=1.0.2-b \
--frequency-plan-id EU_863_870

```

get the session info : 

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices get caf 850706c7689521a801689b5f7a1b4ec5 --session
{
  "ids": {
    "device_id": "850706c7689521a801689b5f7a1b4ec5",
    "application_ids": {
      "application_id": "caf"
    },
    "dev_eui": "A81758FFFE0367C0",
    "join_eui": "70B3D59520000000"
  },
  "created_at": "2019-03-26T12:47:28.028Z",
  "updated_at": "2019-03-26T12:47:28.028Z",
  "network_server_address": "localhost:8884",
  "application_server_address": "localhost:8884",
  "session": {
    "dev_addr": "00000000",
    "keys": {
      "f_nwk_s_int_key": {
      },
      "s_nwk_s_int_key": {
      },
      "nwk_s_enc_key": {
      },
      "app_s_key": {
        "key": "A37hCODOvb3xrVbU6veHFg=="
      }
    },
    "started_at": "0001-01-01T00:00:00Z"
  }
}

```

Step 2: have the device join
===================

after JR:

```
go run ./cmd/ttn-lw-cli  --ca="/Users/pdubois/go/src/go.thethings.network/lorawan-stack/cert.pem" devices get caf 850706c7689521a801689b5f7a1b4ec5 --session
{
  "ids": {
    "device_id": "850706c7689521a801689b5f7a1b4ec5",
    "application_ids": {
      "application_id": "caf"
    },
    "dev_eui": "A81758FFFE0367C0",
    "join_eui": "70B3D59520000000"
  },
  "created_at": "2019-03-26T12:47:28.028Z",
  "updated_at": "2019-03-26T12:47:28.028Z",
  "network_server_address": "localhost:8884",
[... 233 lines cut ...]
Decoded packet
--------------
Message Type = Data
            PHYPayload = 600F6900FC000200068A3CB31220F774
          ( PHYPayload = MHDR[1] | MACPayload[..] | MIC[4] )
                  MHDR = 60
            MACPayload = 0F6900FC000200068A3CB3
                   MIC = 1220F774 (OK)
          ( MACPayload = FHDR | FPort | FRMPayload )
                  FHDR = 0F6900FC000200
                 FPort = 06
            FRMPayload = 8A3CB3
             Plaintext = 3E01FE ('>..')
                ( FHDR = DevAddr[4] | FCtrl[1] | FCnt[2] | FOpts[0..15] )
               DevAddr = FC00690F (Big Endian)
                 FCtrl = 00
                  FCnt = 0002 (Big Endian)
                 FOpts = 
          Message Type = Unconfirmed Data Down
             Direction = down
                  FCnt = 2  <- ** this is WRONG again **
             FCtrl.ACK = false
             FCtrl.ADR = false

```

PR #440 (2019-04-03): Invalidate and validate application downlink queue

**Summary:**

Closes #357 
Closes #414 

**Changes:**

- NS sends queue invalidation on MAC layer downlinks also when there is no application queue. This is to communicate the `AFCntDown`
- NS verifies that the `FCnt` in the dequeued downlink message has the right `FCnt` in 1.0.x; this is a final sanity check before transmission
- NS validates the incoming pushed/replaced application downlinks; that they have an existing session and a valid frame counter

**Notes for Reviewers:**

You may want to review commit by commit.

The case where an un [... cut]
closed 2019-04-03 by johanstokking