TheThingsNetwork/lorawan-stack #385: Compliance with Semantic Import Versioning

htdvisser, 2019-03-28:
**Summary:** 

Our current module path `go.thethings.network/lorawan-stack` is not compliant with semantic import versioning. [Go's March 2019 blog post on modules](https://blog.golang.org/using-go-modules) contains the following:

> Each different major version (v1, v2, and so on) of a Go module uses a different module path: starting at v2, the path must end in the major version.

Considering that we're v3 now, this would mean that our import path should be `go.thethings.network/lorawan-stack/v3`.

**What do you see now?**

We use the import path `go.thethings.network/lorawan-stack`

**What do you want to see instead?**

We should investigate if the current import path will cause any problems. We are compliant with the first part of the quote above, because we don't have a previous major version of the module path. I think the second part might cause problems, because we are higher than v2, so the path _must_ end in the major version.

**Environment:**

Go 1.11+ with 1.11 modules enabled.

**How do you propose to implement this?**

If we decide to change this, we have to change `go.mod` and do a find/replace of:

* `"go.thethings.network/lorawan-stack` to `"lorawan-stack "go.thethings.network/lorawan-stack/v3`
* `"go.thethings.network/lorawan-stack/` to `"go.thethings.network/lorawan-stack/v3/`

**Can you do this yourself and submit a Pull Request?**

Let's discuss it first. If we decide do it, I can submit a PR.

rvolosatovs, 2019-04-15:
I don't see a reason not to do this.
Let's make sure we do this before people start depending on our project - these kind of renames are a royal PITA to sync downstream and people will start getting dependency conflicts.

johanstokking, 2019-04-15:
I’m fine doing this but consider that this project is not intended to be used as library. Our compatibility commitment is not about code. So it’s nice, but `v3` doesn’t say much in terms of compatibility

For the Go SDK, it’s a different story and I think we should consider putting that in a separate repo with it’s own generated code to keep dependencies clean. There we do have a public code API to maintain, which calls for semantic versioning.

htdvisser, 2020-07-14:
@rvolosatovs fixed this in #2453.

closed 2020-07-14 by htdvisser