TheThingsNetwork/lorawan-stack #399: Use time margin for class A Rx window expiration detection

johanstokking, 2019-03-29:
**Summary:** 

The implementation of #389 uses NS server-time check using RX1 delay whether the class A downlink window(s) can be made. This does not account for;

1. Network latency to GS; this may be through PP-PB-PP
2. GS `ScheduleTimeShort` (default 100ms GS-Gtw network latency + 30ms internal gateway queue delay)

Suggesting that NS uses a sane margin to not try the downlink path anymore.

**Why do we need this?**

Trying downlink paths that are very likely to be expired when the Tx comes at the gateway costs scheduling time due to potential high latencies, making other options impossible, resulting eventually in missing all opportunities.

**What is already there? What do you see now?**

No margin; NS server-time based on Rx delay.

**What is missing? What do you want to see?**

A margin. This would be bare minimum the GS' `ScheduleTimeShort`. We can have three phases of implementing this;

**How do you propose to implement this?**

1. MVP; use NS hardcoded `ScheduleTimeShort`; don't account for NS-GS delay (fine for private networks in same network)
2. Measure roundtrip time between NS and the GS instance (some application level ping/pong). Consider this part of the cluster / peering (cc @htdvisser) and add to that
3. Once GS knows the specific gateway roundtrip time (per https://github.com/TheThingsNetwork/lorawan-stack/issues/398), consider using that. This does mean, however, that NS gets a dictionary of all gateways and their roundtrip times. This may not be desirable, so keep in mind;
   - do not store in memory but shared cache across NS instances (i.e. Redis registry)
   - limits (i.e. Redis key TTL per gateway)

**Can you do this yourself and submit a Pull Request?**

Will review.

htdvisser, 2019-04-01:
> Measure roundtrip time between NS and the GS instance (some application level ping/pong). Consider this part of the cluster / peering (cc @htdvisser) and add to that

#418

johanstokking, 2020-03-25:
Replaced by #2078

closed 2020-03-25 by johanstokking