TheThingsNetwork/lorawan-stack #414: MIC check fails when sending application downlink after MAC downlink

johanstokking, 2019-03-31:
**Summary:** 

MIC check on the end device fails when an application downlink is queued after sending MAC layer downlink.

**Steps to Reproduce:** 

1. Join device in `1.0.2`
2. Have some MAC layer downlink happen (i.e. confirmed up, no app down)
3. Enable class C
4. Send an application downlink message

**What do you see now?**

MIC check fails.

**What do you want to see instead?**

Successful downlink.

**How do you propose to implement this?**

- Currently, when there's application payload queued, NS sends queue invalidation. This should also happen when there's no application payload queued, so that AS has the right `NFCntDown`
- There should also be a check in NS of application downlink that uses a wrong counter

**Can you do this yourself and submit a Pull Request?**

Yes, probably

pdubois22, 2019-03-31:
related to #357 ?

johanstokking, 2019-04-01:
@pdubois22 very likely the same thing indeed.

PR #440 (2019-04-03): Invalidate and validate application downlink queue

**Summary:**

Closes #357 
Closes #414 

**Changes:**

- NS sends queue invalidation on MAC layer downlinks also when there is no application queue. This is to communicate the `AFCntDown`
- NS verifies that the `FCnt` in the dequeued downlink message has the right `FCnt` in 1.0.x; this is a final sanity check before transmission
- NS validates the incoming pushed/replaced application downlinks; that they have an existing session and a valid frame counter

**Notes for Reviewers:**

You may want to review commit by commit.

The case where an un [... cut]
closed 2019-04-03 by johanstokking