TheThingsNetwork/lorawan-stack #419: Blacklist console routes entries

bafonins, 2019-04-01:
**Summary:** 

It is possible to create an entity with the `add` as an identifier. This causes issues in console routing, since we use `/console/<entity-id>/add`  for the entity add page and hence the entity is not reachable.

**Steps to Reproduce:** 

1. Create a new application with `id = add`
2. Navigate to `/console/applications/add`
3. See the Add Application page instead of the Application Overview page for the `add` application. 

**What do you see now?**

The Add Application page

**What do you want to see instead?**

The Application Overview page

**Environment:**

is, browser

**How do you propose to implement this?**

Extend the blacklist 

**Can you do this yourself and submit a Pull Request?**

Fix the issue

htdvisser, 2019-04-01:
Let's do this before release, so that we don't risk users already adding those.

After looking through the current list, I think we should add the following IDs to the blacklist:

- `add`
- `edit`
- `device`+`devices`
- `end_device`+`end_devices`
- `collaborator`+`collaborators`

Maybe @johanstokking or @KrishnaIyer has more additions

johanstokking, 2019-04-01:
Good catch, and classic problem that easily overseen.

I think we shouldn't be too permissive with IDs to not get in trouble later. This spans from forbidden names, to confusion, to pretending it's some builtin or system thing, to url collisions, to things we want to freely use in examples (i.e. `yourapp`).

Here's a list that's being referenced a lot; https://github.com/shouldbee/reserved-usernames/blob/master/reserved-usernames.json (other formats available).

bafonins, 2019-04-01:
Shall we include all of those?

johanstokking, 2019-04-01:
Better safe than sorry; I'd do it.

PR #420 (2019-04-02): Extend IS blacklist

**Summary:**

Closes #419 

[image]

**Changes:**

- Extend IS blacklist with entries from https://github.com/shouldbee/reserved-usernames/blob/master/reserved-usernames.json
- Fix `<Form />` error to field mapping
- Extend the Application Add form with the `blacklist_id` error

**Notes for Reviewers:**

@htdvisser please check the blacklist extension
@kschiffer please check the console related changes

**Release Notes: _(optional)_**
closed 2019-04-02 by bafonins