TheThingsNetwork/lorawan-stack #422: Client side pagination for api-keys

bafonins, 2019-04-01:
**Summary:** 

There is no server side pagination for API Keys, however the console relies on this fact.
Note that only gateways, applications, devices and organizations are paginated on the server side.

**Steps to Reproduce:** 

1. Go to the API Keys List page
2. Exceed the page size
3. Observe no pagination

**What do you see now?**

API keys are not paginated

**What do you want to see instead?**

API keys being paginated

**Environment:**

browser

**How do you propose to implement this?**

Simply add pagination to the `<Tabular />` component

**Can you do this yourself and submit a Pull Request?**

Fix it

htdvisser, 2019-04-02:
This might be difficult. The request messages for `ListAPIKeys` RPCs don't support pagination in the backend.

bafonins, 2019-04-02:
@htdvisser thats why the PR for this issue introduces pagination on the client side

htdvisser, 2019-04-02:
Ok, that wasn't clear (at least not to me) from the original issue in itself. The combination of

> There is no server side pagination for API Keys, however the console relies on this fact.

and

> **What do you want to see instead?**
> API keys being paginated

gave me the impression that you did want this, hence my comment.

PR #423 (2019-04-02): Fix API Keys pagination

**Summary:**

Closes #422 
Related to https://github.com/TheThingsNetwork/lorawan-stack/issues/268  as well

**Changes:**

- Add client side pagination to the `<Tabular />` component
closed 2019-04-02 by bafonins