TheThingsNetwork/lorawan-stack #439: Get LastNFCntDown from NS in 1.0.x before recalculating queue

johanstokking, 2019-04-02:
**Summary:** 

Get `LastNFCntDown` from NS in 1.0.x before recalculating the downlink queue.

**Why do we need this?**

In case AS lost link and downlink was sent by NS, the AS' `LastAFCntDown` is wrong. Since in 1.0.x this needs to stay in sync with `LastFCntDown`, this value should be fetched from NS first.

**What is already there? What do you see now?**

Recalculation and validation of pushed/replaced downlinks. Since this validation fails, the downlink queue is cleared.

Also, the AS only gets the right `LastAFCntDown` value from NS when NS emits a MAC layer downlink or when NS invalidates the downlink queue.

**What is missing? What do you want to see?**

When AS pushes or replaces new application downlink with a frame counter that is too low and in 1.0.x, it should get the `LastNFCntDown` from NS and retry.

**How do you propose to implement this?**

On push/replace, AS needs a way to determine that the error is `fcnt_too_low`. This is currently `InvalidArgument`, but `InvalidArgument` can be another validation error. Still, AS may retry with this error code and fail on the second failure.

Ideas @htdvisser ?

**Can you do this yourself and submit a Pull Request?**

Yes

htdvisser, 2019-04-03:
I think the NS should just queue `ApplicationUp` for the AS while there is no link. When the link is back, the queue is flushed and the AS will receive all `downlink_queue_invalidated` ups, and be able to recalculate.

rvolosatovs, 2019-04-15:
@htdvisser so you propose to implement a per-AS uplink queue in NS stored in a registry?
A runtime queue would not work, since all the uplinks would be lost once a NS instance dies

htdvisser, 2019-04-16:
Correct

johanstokking, 2019-04-23:
Replaced by #560

rvolosatovs, 2019-04-23:
@johanstokking so can we close this one?

closed 2019-04-23 by htdvisser