TheThingsNetwork/lorawan-stack #441: DevEUIs are not unique in Identity Server

htdvisser, 2019-04-02:
**Summary:** 

We may have to add a "unique" constraint to the DevEUI field of EndDevice.

**What do you see now?**

The Network Server and Join Server seem to assume that DevEUIs are unique (which per spec they should be). However, the Identity Server only has a "unique" constraint on the combination of JoinEUI+DevEUI. 

**What do you want to see instead?**

I think we should be fully compliant and treat DevEUIs as unique. 

**How do you propose to implement this?**

If @johanstokking agrees it's a simple change from `index` to `unique_index` in `pkg/identityserver/store/end_device.go`.

**Can you do this yourself and submit a Pull Request?**

Sure

johanstokking, 2019-04-02:
The requirement that we set is that V3 stack is compliant with LoRaWAN and the Backend Interfaces. Both do not enforce uniqueness, but there are a few things to note.

First, the device registry in all components (IS, JS, NS, AS) do enforce uniqueness of the combination of JoinEUI and DevEUI.

Second, the Join Server key registry only takes DevEUI into account, because considering Backend Interfaces of JS, we only have DevEUI. However, we should ensure that the session key ID does not collide across multiple devices with a different JoinEUI but same DevEUI.

The use cases that we want to be supporting are:

1. Users use the device serial number or some chip ID as DevEUI. These should not collide with "real" DevEUIs (i.e. from a IEEE issued block). Otherwise, in PCN, we get weird behavior.
2. Users use the device serial number but migrate from one JS to another JS, without changing the DevEUI. This migration may result in a transfer of ownership, which means recreation of the device in a different application, keeping the DevEUI.

So I think we should stay with the current behavior and at this point I don't see a need for a change.

htdvisser, 2019-04-03:
Doesn't the definition of Extended Unique Identifier (EUI) alone already enforce uniqueness? At least for universally administered addresses (UAAs) it enforces universal uniqueness. For locally administered addresses (LAAs) we can indeed specify that the address is unique within the namespace of a join server.

If we're supporting duplicate DevEUIs in our NS/JS, how will we handle RejoinRequests of type 0 and type 2?

johanstokking, 2019-04-03:
>If we're supporting duplicate DevEUIs in our NS/JS, how will we handle RejoinRequests of type 0 and type 2?

MIC check:

[image]

We have considered use cases, practical implications and technical compatibility with LoRaWAN and Backend Interfaces and we there are no issues to my understanding. In fact, TC considers JS to JS migration in different ways, all of them involve changing the JoinEUI but not (necessarily) the DevEUI. Again, see use case 2 mentioned above.

htdvisser, 2019-04-03:
Ok

closed 2019-04-03 by htdvisser