TheThingsNetwork/lorawan-stack #444: Fix button props propagation

bafonins, 2019-04-02:
**Summary:** 

https://github.com/TheThingsNetwork/lorawan-stack/pull/296 broke the `<Button />` component; it ignores the `disabled` state. Please check for other issues related to the component

**Steps to Reproduce:** 

1. Open any form that has validation
2. Fail validation
3. Submit button is not disabled

**What do you see now?**

No `disabled` state on the `<Button />` component.

**Environment:**

browser

**How do you propose to implement this?**

Resolve component props properly

**Can you do this yourself and submit a Pull Request?**

review

bafonins, 2019-04-02:
Besides that there is also unused `link` prop

kschiffer, 2019-04-03:
Will fix. Handling of html props is always pretty wonky. Maybe it's time to use something like [react-html-attributes](https://www.npmjs.com/package/react-html-attributes) to filter html props properly instead of either blindly spreading input props or building prop filters ourselves.

PR #446 (2019-04-03): Fix Button HTML props propagation
**Summary:**
Closes #444.

**Changes:**
- Pass relevant html props to the <button> element
- Remove unnecessary `link` prop
- Update proptypes definitions

**Notes for Reviewers:**
I deliberately left out html props that we will likely not pass. We can add them whenever the demand is there.
closed 2019-04-03 by kschiffer