TheThingsNetwork/lorawan-stack #466: Docker image binds port 11885

adriansmares, 2019-04-04:
**Summary:** 

The docker image binds port `11885` even though this port is no longer used. This results in the CLI no longer being able to login, since the bind will fail.

**Steps to Reproduce:** 

1. Start the docker container
2. Attempt to `ttn-lw-cli login`

Not necessary.

**What do you see now?**

```
 ERROR Could not listen for OAuth callback. Try re-running this command with --callback=false error=listen tcp :11885: bind: address already in use
listen tcp :11885: bind: address already in use
exit status 255
```

**What do you want to see instead?**

No error.

**Environment:**

Latest master.

**How do you propose to implement this?**

Remove the port `11885` bind from the `Dockerfile`.

**Can you do this yourself and submit a Pull Request?**

Yes.

KrishnaIyer, 2019-04-04:
The port indeed needs to be removed. But have you tested with the newest CLI? 
https://github.com/TheThingsNetwork/lorawan-stack/pull/433
It uses a local callback and won't redirect to `11885`.

adriansmares, 2019-04-04:
Closed by #467

PR #467 (2019-04-04): util: Remove docker-compose unused bind

**Summary:**

Closes #466 

**Changes:**

- Remove port `11885` from `docker-compose.yml`

**Notes for Reviewers:**

None

**Release Notes: _(optional)_**

None
closed 2019-04-04 by adriansmares