TheThingsNetwork/lorawan-stack #488: CLI does not inform user about logout on OAuth Server

htdvisser, 2019-04-08:
**Summary:** 

On logout, the CLI doesn't inform the user that they're only logging out of the CLI and that they should still logout of the OAuth server. This makes switching accounts difficult.

Originally reported by @johanstokking 

**Steps to Reproduce:** 

1. `ttn-lw-cli login`
2. Log in to the OAuth server and authorize the CLI
3. Assuming the user wants to login with a different account, run `ttn-lw-cli logout` and `ttn-lw-cli login` again.

**What do you see now?**

Since the user already authorized the CLI, they are redirected immediately.

**What do you want to see instead?**

They should not be redirected immediately. Couple of options for this:

**How do you propose to implement this?**

Option 1: we can show an account switcher view like Google does:

[image]

After optionally switching accounts, we follow the normal flow.

Option 2: after successful authorization, the CLI redirects the user back to a "success" page on the OAuth server, where the user can then log themselves out of the OAuth server if they want. Added benefit is that we can render a nice page instead of a white "you are now logged in" page.

We can even do a combination of the two, depending on how much work we want to do.

**Can you do this yourself and submit a Pull Request?**

Let's discuss how we want to approach this. After that I can implement it.

johanstokking, 2019-04-08:
I'd go for option 1.

kschiffer, 2019-04-29:
Option 1 is certainly better UX wise. 
Option 2 would be quicker though and we also already have a (minimal) page to logout of the OAuth server: `/oauth`. So, maybe it's good to start with this and add the account chooser later, depending on how urgent this is.

htdvisser, 2019-06-25:
As discussed, we'll go with the account switcher. Until then we can consider redirecting users to a /oauth/logout page when they logout of the cli/console

johanstokking, 2021-03-01:
@kschiffer please open a new issue with progressive insight with the account app.

closed 2021-03-01 by johanstokking