TheThingsNetwork/lorawan-stack #505: Reset MAC state on ABP session FCnt reset

johanstokking, 2019-04-08:
**Summary:** 

Reset the MAC state when the FCnt resets (and it's permitted, of course).

**Why do we need this?**

Because the device has power cycled and doesn't have a local MAC state anymore.

**What is already there? What do you see now?**

MAC state resets when the device confirms the pending session.

**What is missing? What do you want to see?**

There should be similar behavior when the device "confirms" an ABP session reset.

**Environment:**

`v3.0.0`

**How do you propose to implement this?**

Make the `resetMACState()` part of the uplink flow when a counter reset has been detected.

**Can you do this yourself and submit a Pull Request?**

Yes, can review too. Will see if I have time.

htdvisser, 2019-04-09:
And only on LoRaWAN 1.0.x, because 1.1+ devices send the ResetInd command.

johanstokking, 2019-04-09:
`ResetInd` is indeed for 1.1 ABP devices, but this is about the entire MAC state except frame counters;

[image]

This issue is about the behavior of `resets_f_cnt`, which is a flag which escapes the normative spec, especially in 1.1. So contrary of `ResetInd`, this `resets_f_cnt` resets the frame counters, also in 1.1. We may print warnings or disable this entire feature NS instance wide through config, but that's also out of scope here.

closed 2019-07-03 by rvolosatovs

BenjiAmi, 2025-01-27:
Hello,

It seems the issue aims to automate the MAC state reset when a device frame counter reset is detected. However, the provided solution (resetMACState() function) doesn't appear to achieve this.

I’m encountering a problem with ABP devices and MAC state: a frame counter mismatch occurred on one of my ABP sensors without a power cycle.

Here are my questions:

1. Has the resetMACState() function been implemented? If so, how does it detect a counter reset and trigger the function? (Reference: @johanstokking - "Make the resetMACState() part of the uplink flow when a counter reset has been detected.")
2. How can The Things Network CLI be used to reset the MAC state of a specific sensor?

PR #858 (2019-07-02): Uplink MAC state handling fixes

#### Summary

Closes #505 
Closes #795 
~References #42 (closes NS part)~
References #789 (closes uplink part)
Rerences https://github.com/TheThingsNetwork/lorawan-stack/issues/560
~Closes https://github.com/TheThingsNetwork/lorawan-stack/issues/23~ (https://github.com/TheThingsNetwork/lorawan-stack/pull/860)

#### Changes

- Derive MAC state(apply MAC commands, account for MAC reset) of the device before computing MIC
- ~Add testing utilities for events~
- ~Reimplement NS uplink testing~
- ~Harmonize NS testing~
- Do not set `ReceivedAt` in AS uplink(that's a  [... cut]