TheThingsNetwork/lorawan-stack #510: Publish Docker images for Arm

johanstokking, 2019-04-09:
**Summary:** 

Publish Docker images for Arm.

**Why do we need this?**

For those wanting to run the stack using Docker on Arm systems, i.e. RPi and Arm servers.

**What is already there? What do you see now?**

Only AMD64 Linux

**What is missing? What do you want to see?**

Also Armv6 and Armv7 32-bit as well as Armv8 64-bit Linux.

**Environment:**

`v3.0.0` on Arm based systems

**How do you propose to implement this?**

Add to `.goreleaser.yml`

**Can you do this yourself and submit a Pull Request?**

Yes

johanstokking, 2019-04-09:
It looks like, in order to do this in a clean way, there needs to be Docker manifest support in Goreleaser.

htdvisser, 2019-09-17:
We could start by adding the following to Goreleaser:

```diff
diff --git a/.goreleaser.yml b/.goreleaser.yml
index 1cbb31cb5..e90b9bcbb 100644
--- a/.goreleaser.yml
+++ b/.goreleaser.yml
@@ -176,3 +176,44 @@ dockers:
     skip_push: auto
     extra_files:
       - public
+  - goos: linux
+    goarch: arm64
+    binaries:
+      - ttn-lw-cli
+      - ttn-lw-stack
+    image_templates:
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm64v8-linux:latest'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm64v8-linux:{{ .Major }}'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm64v8-linux:{{ .Major }}.{{ .Minor }}'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm64v8-linux:{{ .Version }}'
+    skip_push: auto
+    extra_files:
+      - public
+  - goos: linux
+    goarch: arm
+    goarm: 7
+    binaries:
+      - ttn-lw-cli
+      - ttn-lw-stack
+    image_templates:
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v7-linux:latest'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v7-linux:{{ .Major }}'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v7-linux:{{ .Major }}.{{ .Minor }}'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v7-linux:{{ .Version }}'
+    skip_push: auto
+    extra_files:
+      - public
+  - goos: linux
+    goarch: arm
+    goarm: 6
+    binaries:
+      - ttn-lw-cli
+      - ttn-lw-stack
+    image_templates:
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v6-linux:latest'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v6-linux:{{ .Major }}'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v6-linux:{{ .Major }}.{{ .Minor }}'
+      - '{{ or (index .Env "DOCKER_IMAGE") "lorawan-stack" }}-arm32v6-linux:{{ .Version }}'
+    skip_push: auto
+    extra_files:
+      - public
```

Then at least we're publishing Docker images that can be used on ARM. 

We can publish the manifest by hand, or wait for Goreleaser to support it.

johanstokking, 2019-09-17:
Fine with me.

neoaggelos, 2020-07-23:
Picking this one up

closed 2021-03-30 by htdvisser