TheThingsNetwork/lorawan-stack #520: linux386_stack_1 exited with code 255

ramesh3443, 2019-04-10:
**Summary:** 

All the steps are followed as per the guide  https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/gettingstarted.md#dependencies
...

**Steps to Reproduce:** 

Start docker-compose up
the result is as follows 
Starting lorawanstack300linux386_cockroach_1 ...
Starting lorawanstack300linux386_redis_1 ...
Starting lorawanstack300linux386_cockroach_1
Starting lorawanstack300linux386_redis_1 ... done
Starting lorawanstack300linux386_stack_1 ...
Starting lorawanstack300linux386_stack_1 ... done
Attaching to lorawanstack300linux386_cockroach_1, lorawanstack300linux386_redis_1, lorawanstack300linux386_stack_1
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | *
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | *
cockroach_1  | * Check out how to secure your cluster: https://www.cockroachlabs.com/docs/v2.1/secure-a-cluster.html
cockroach_1  | *
redis_1      | 1:C 10 Apr 2019 12:06:00.143 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 10 Apr 2019 12:06:00.143 # Redis version=5.0.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 10 Apr 2019 12:06:00.143 # Configuration loaded
redis_1      | 1:M 10 Apr 2019 12:06:00.148 * Running mode=standalone, port=6379.
redis_1      | 1:M 10 Apr 2019 12:06:00.148 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1      | 1:M 10 Apr 2019 12:06:00.148 # Server initialized
redis_1      | 1:M 10 Apr 2019 12:06:00.148 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
redis_1      | 1:M 10 Apr 2019 12:06:00.148 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1      | 1:M 10 Apr 2019 12:06:00.148 * Ready to accept connections
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "355ee9827b18" to other nodes, is this routable?
cockroach_1  | *
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | *
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2019-04-10 12:06:00.143450224 +0000 UTC (took 0.8s)
cockroach_1  | build:               CCL v2.1.1 @ 2018/11/19 18:24:21 (go1.10.3)
cockroach_1  | webui:               http://355ee9827b18:26256
cockroach_1  | sql:                 postgresql://root@355ee9827b18:26257?sslmode=disable
cockroach_1  | client flags:        /cockroach/cockroach <client cmd> --host=355ee9827b18:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp238173095
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | status:              restarted pre-existing node
cockroach_1  | clusterID:           1b00d56c-0b81-49ff-bfda-0cec57563a00
cockroach_1  | nodeID:              1
stack_1      |   INFO Setting up core component
stack_1      |   WARN No cookie hash key configured, generated a random one hash_key=A5F221342879C3C2E13BABEC80A4DB2371D1DDEA5A7E1176915152F350727A45AF8D16EB4B38D24A7D10FCA6BD474B2D40DEDF9659B3EDEE197292E6DD708F1C namespace=web
stack_1      |   WARN No cookie block key configured, generated a random one block_key=F8A6875F83DAC578217B78A4843ED701CAE02375C66065C4CF75BD55140361E9 namespace=web
stack_1      |   INFO Setting up Identity Server
stack_1      |   INFO Setting up Gateway Server
stack_1      | error:cmd/internal/shared:initialize_gateway_server (could not initialize Gateway Server)
stack_1      | --- error:pkg/gatewayserver:listen_frontend (failed to start frontend listener `tls` on address `:8882`)
stack_1      |     protocol=tls
stack_1      |     address=:8882
stack_1      | --- read /run/secrets/cert.pem: is a directory
lorawanstack300linux386_stack_1 exited with code 255

 --as.link-mode="all"
                                                --as.mqtt.listen=":1883"
[... 150 lines cut ...]
**What do you see now?**

...

**What do you want to see instead?**

...

**Environment:**

Version:             3.0.0
Go version:          go1.11.6
OS/Arch:             linux/amd64

...

**How do you propose to implement this?**

Some suggestions from the developer

...

**Can you do this yourself and submit a Pull Request?**

...

KrishnaIyer, 2019-04-10:
@ramesh3443: The error message is quite clear here:
`error:pkg/gatewayserver:listen_frontend (failed to start frontend listener tls on address :8882`.

Can you check if port `:8882` is already bound to another process?

ramesh3443, 2019-04-10:
Hi KrishnaIyer, 

I dont see any process running 8882
Please see the attached screenshot
[image]

johanstokking, 2019-04-10:
@ramesh3443 the mount is wrong;

```
stack_1 | --- read /run/secrets/cert.pem: is a directory
```

ramesh3443, 2019-04-10:
@johanstokking Can you suggest what went wrong, what should i do.

johanstokking, 2019-04-10:
@ramesh3443 make sure you have a file `cert.pem` and `key.pem` in the local directory that is readable by the Docker user `886`

Please let us know if we need to clarify something in the Getting Started.

ramesh3443, 2019-04-10:
Hi @johanstokking 
The server is now running 

redis_1      | 1:M 10 Apr 2019 13:05:16.236 * Ready to accept connections
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | *
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | *
cockroach_1  | * Check out how to secure your cluster: https://www.cockroachlabs.com/docs/v2.1/secure-a-cluster.html
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "df60cf7fab12" to other nodes, is this routable?
cockroach_1  | *
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | *
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2019-04-10 13:05:20.653605857 +0000 UTC (took 2.6s)
cockroach_1  | build:               CCL v2.1.1 @ 2018/11/19 18:24:21 (go1.10.3)
cockroach_1  | webui:               http://df60cf7fab12:26256
cockroach_1  | sql:                 postgresql://root@df60cf7fab12:26257?sslmode=disable
cockroach_1  | client flags:        /cockroach/cockroach <client cmd> --host=df60cf7fab12:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp067124035
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | status:              initialized new cluster
cockroach_1  | clusterID:           83ecd71e-1d38-4873-b54b-a3c30ec8a4ab
cockroach_1  | nodeID:              1
stack_1      |   INFO Setting up core component
stack_1      |   WARN No cookie hash key configured, generated a random one hash_key=FB6E2250E624E4977ED260CD3114EF4CD9EC91517FC18B37E9604D3BE0B420F2E07B6244C1B7D93B22252B70474CC0B0DEB3951918F5E2F061F3748ABE715E10 namespace=web
stack_1      |   WARN No cookie block key configured, generated a random one block_key=CD013A37709443962A62ED2044B3F6B5C0843F27B6026B5850357BC88130D101 namespace=web
stack_1      |   INFO Setting up Identity Server
stack_1      |   INFO Setting up Gateway Server
stack_1      |   INFO Setting up Network Server
stack_1      |   INFO Setting up Application Server
stack_1      |   INFO Setting up Join Server
stack_1      |   INFO Starting...
stack_1      |   INFO Listening for gRPC connections           address=:1884 namespace=grpc
stack_1      |   INFO Listening for gRPC/tls connections       address=:8884 namespace=grpc
stack_1      |   INFO Listening for HTTP connections           address=:1885 namespace=web
stack_1      |   INFO Listening for HTTPS connections          address=:8885 namespace=web
stack_1      |   WARN No cluster key configured, generated a random one key=00d4fac9fab8b56085c594f38a14aede955429799d43e889578fe0330792ce2f

But on the other terminal when i run the command ttn-lw-cli login,

it is giving following error
Command 'ttn-lw-cli' is available in '/snap/bin/ttn-lw-cli'
The command could not be located because '/snap/bin' is not included in the PATH environment variable.
ttn-lw-cli: command not found
root@rgserver:/home/rg/lorawan-stack_3.0.1_linux_386#

ramesh3443, 2019-04-10:
back=false error=listen tcp :11885: socket: permission denied
listen tcp :11885: socket: permission denied

ramesh3443, 2019-04-10:
[image]

johanstokking, 2019-04-10:
@ramesh3443 please see your OS instructions to learn how to add directories to `PATH`.

As for the listen error; you pass `--callback=false` if you don't have permissions to start a listener.

ramesh3443, 2019-04-10:
Hi @johanstokking 

Please see the error after entering the url code into the cli
[image]

Key code MF2XI.JT73MDRBVZ42MWJ2NXLBWUFLUAYUXXFROM644MQ.YS6URLJ7YQ25JLT7LHFRH4A6RMDAFBDL2ZXUW4TS4AGRGWBFBX3Q

johanstokking, 2019-04-10:
@ramesh3443 looks like your current user is underprivileged to even do a DNS lookup.

What environment is this? What kind of user is this?

Can you try with [`sudo`](https://kb.iu.edu/d/amyi)?

ramesh3443, 2019-04-10:
Hi @johanstokking 

I am running on Ubuntu 18.04 one is on my local VMware machine and other instance on AWS, i am getting same error.
I have tried already with SUDO on both machines

johanstokking, 2019-04-10:
OK, this is indeed an issue, see https://github.com/TheThingsNetwork/lorawan-stack/issues/523

johanstokking, 2019-04-15:
Closed by #523

closed 2019-04-15 by johanstokking