TheThingsNetwork/lorawan-stack #536: RejoinRequest simulation

htdvisser, 2019-04-16:
**Summary:** 

The CLI is getting commands for simulating LoRaWAN traffic, but currently only joins and (un)confirmed data messages are supported. We should add support for rejoins as well.

**Why do we need this?**

This will help with debugging our stack.

**What is already there? What do you see now?**

The `ttn-lw-cli simulate` commands for `uplink` and `join-request`.

**What is missing? What do you want to see?**

Another one for `rejoin-request`

**How do you propose to implement this?**

Copy join request simulation, but with different fields and different crypto.

**Can you do this yourself and submit a Pull Request?**

Sure, but not a priority right now. Feel free to re-assign this issue if you want to work on this.

htdvisser, 2020-08-24:
Let's do this when we need it. We haven't even seen the first LoRaWAN 1.1 device yet.

neoaggelos, 2021-08-18:
Removing milestone, no point in doing this before Rejoins are actually supported. https://github.com/TheThingsNetwork/lorawan-stack/issues/8