TheThingsNetwork/lorawan-stack #544: NS send linkadrReq to open 48 uplink channel in CN470 region

yunfei83, 2019-04-16:
**Summary:** 

NS server  send linkadrReq  frame  to open device 48 uplink channel in CN470 region

**Steps to Reproduce:** 

Step1:  the device  successfully joined  the network
Step2: Device Send first data length（33byte）:<40 69 3D F3 01 80 00 00 0A 29 75 1B CD C5 DE 90 DB D0 06 CD 9E 89 66 5A 7A 9F 0A 91 FA 8A F8 3D 31 >
Step3: NetworkServer responses linkadrReq mac command 

**What do you see now?**

linkadrReq (open 48 uplink channel: 0-7, 16-23, 32-39, 48-55, 64-71, 80-87)
 03 00 FF 00 01 
 03 00 FF 00 11 
 03 00 FF 00 21 
 03 00 FF 00 31 
 03 00 FF 00 41 
 03 00 FF 00 51 
 06

after joined : 
mac-state.current-parameters.channels  = 96 channels enable
mac_state.desired_parameters.channels = 8 channels enable
after send data：
mac-state.current-parameters.channels  = 96 channels enable
if success respond  linkadrans(0307...), mac-state.current-parameters.channels  will change to  8 channels enable
mac_state.desired_parameters.channels = 8 channels enable

**What do you want to see instead?**

only open 8 channels 
>
 03 00 FF 00 01 
 03 00 00 00 11 
 03 00 00 00 21 
 03 00 00 00 31 
 03 00 00 00 41 
 03 00 00 00 51 

**Environment:**

>
The Things Network Stack for LoRaWAN: ttn-lw-stack
Version:             v3.0.0-pre-dev
Go version:          go1.11.5
OS/Arch:             linux/amd64

**How do you propose to implement this?**

...

**Can you do this yourself and submit a Pull Request?**

...

htdvisser, 2019-04-17:
@rvolosatovs can you look into this? If you can confirm the issue, please assign the `bug` label and a milestone.

rvolosatovs, 2019-04-17:
@yunfei83 can you please try version `v3.0.2`(https://github.com/TheThingsNetwork/lorawan-stack/releases/tag/v3.0.2) and confirm that issue happens there too?

yunfei83, 2019-04-17:
> @yunfei83 can you please try version `v3.0.2`(https://github.com/TheThingsNetwork/lorawan-stack/releases/tag/v3.0.2) and confirm that issue happens there too?

3.0.2 has the same problem

yunfei83, 2019-04-17:
LinkadrReq downlink chmask is not mac_state.desired_parameters.channels , I try add a new freq plan id ,  show the same problem.

htdvisser, 2019-04-23:
This problem was fixed by https://github.com/TheThingsNetwork/lorawan-frequency-plans/commit/338ffc0e2a4fec5bca2316032412e4dd1365ebf4#diff-be64fca95b8e26b52fd858b48524e7c3

closed 2019-04-23 by htdvisser