TheThingsNetwork/lorawan-stack #545: Data pushed by mobile gateways should be excluded from the ADR algorithm

dmamalis, 2019-04-17:
This is a **feature request** for **the backend**.
 
We plan to install some moving gateways. These gateways can assist in covering hard to reach areas. However, when they move within coverage of other gateways they might cause issues to the ADR algorithm. 

Should we have a way to mark gateways as moving gateways? Could messages being uploaded by the moving gateways be flagged in a way so that they are excluded from the ADR algorithm?

htdvisser, 2019-04-17:
**How do you propose to implement this?**

- Gateways are registered with an extra flag to exclude them from ADR
- This flag is propagated on uplink Rx metadata
- Uplink metadata with this flag is not considered for ADR

htdvisser, 2019-04-17:
Additionally, if uplinks are received on *only* "adr excluded" gateways, the NS should unset the downlink ADR bit.

dmamalis, 2019-04-17:
There is something more although it might not be of much importance:
How do we handle the downlink when a moving gateway moves within vicinity of normal gateways?
If a node is being received by both moving and stationary gateways, the case might be that the moving gateway has recorded better RSSI/SNR than the stationary ones. Shouldn't we somehow make sure that the downlink won't be routed through a moving gateway cause...well...it might not be there anymore?

htdvisser, 2019-04-17:
Indeed, mobile gateways shouldn't be considered for Class C downlinks either. I suppose that they are okay for Class A downlinks, because those are scheduled for max 15 seconds in the future, and I don't think gateways can move so fast that they have the best signal for an uplink, and are out of range 15 seconds later.

dmamalis, 2019-04-17:
Great notice! Class C is indeed the most important!.
In our use case gateways will definitely move slow however in dense city environment and/or if people install them on cars the scenario might not be that rare.

johanstokking, 2019-04-25:
@dmamalis we also have a downlink path contraint for this, which can be "never", "prefer other" or "none". For mobile gateways, GS should override to "prefer other", i.e. indicating that fixed location gateways are tried first regardless of signal quality.

The reason is that we cannot really say "well, the gateway moves slow, so it never impacts class A downlink"; a gateway can move fast enough (high speed train) for class A to influence signal quality selection (especially with high Rx1 delay), which is one of the reasons why we have "prefer other".

Extending labels as we need an indication in `Gateway` and GS has to send it upstream.