TheThingsNetwork/lorawan-stack #551: In getting started，how configure the CLI to trust the self-signed certificates?

enric-lee, 2019-04-18:
Keep in mind that self-signed certificates are not trusted by browsers and operating systems, resulting in warnings and errors such as certificate signed by unknown authority or ERR_CERT_AUTHORITY_INVALID. In most browsers you can add an exception for your self-signed certificate. **You can configure the CLI to trust the certificate as well.**

Can you add how-to about configuring the CLI to trust the self-signed certificates?

enric-lee, 2019-04-18:
The CLI could not exchange the OAuth access token: Post https://localhost:8885/oauth/token: x509: certificate signed by unknown authority.

htdvisser, 2019-04-18:
This issue does not follow the issue template. Please fill the template when submitting issues.

You may want to continue reading the getting started document. The section immediately after the text you posted is called "configuration".

closed 2019-04-18 by htdvisser