TheThingsNetwork/lorawan-stack #553: Events presentation in the console

bafonins, 2019-04-18:
**Summary:** 

We should identify events types that we want to show in the console as well as their content to be presented to the user.
Currently there are two places where we plan to display the latest events related to an entity:
1. On the entity overview page - small view the latest 5-10 events. In this widget view events cannot be expanded and information provided is very limited.
[image]
In order to get more descriptive information related to some specific event or to see the rest of them, the user can navigate through the widget or the side navigation entry to an entity data page which is planned to similar to the Traffic page in the V2 console.

2. The entity data page shows all the events (related to the entity) that are available on the client as well as allows the user expand events to inspect important data included in the event. Filtering will be added later.

**Why do we need this?**

We want to display entity events in the console as well as identify important information related to each event type.

**What is already there? What do you see now?**

1. Events API
2. Events API in the js sdk - https://github.com/TheThingsNetwork/lorawan-stack/pull/522
3. `<List />` and `<VirtualizedList />` components
4. Multiple placehoders in entity overview pages for the `<EventsWidget />` component.
5. `<EventsWidget />` component
6. `<ConnectionStatus />` component

**What is missing? What do you want to see?**

Answers to these questions:
* Can we aggregate events into groups that share similar logic/result/intent in the stack? For example the create/update/delete events can grouped together and hence can be repsented by a single component with varying entries of icon/time and message.
* What are the events we are interested in the most at the moment?
* What data do we want to show (in the expand view) for these events?

**Environment:**

Browser

**How do you propose to implement this?**

A generic `<Event />` component that can be adjusted to different event types, say` the ones that simply inform the user about a certain change and events that include important data like payload/addresses/etc. that can be interesting to the user.

**Can you do this yourself and submit a Pull Request?**

Implement events in the console.

cc @kschiffer @johanstokking @htdvisser @Sypheos

htdvisser, 2019-04-18:
For MVP I'd say we should really start simple.

CRUD events in IS triggered by API actions:

- `<entity>.create`
- `<entity>.update`: the payload is the list of fields being updated
- `<entity>.api-key.create/update/delete`
- `<entity>.collaborator.update/delete`
- `<entity>.delete`

Gateway events in GS:

- `gs.gateway.connect`
- `gs.gateway.disconnect`
- `gs.status.receive`

For these, the `time`, `id` and `event description` fields should be sufficient.

Sypheos, 2019-04-18:
The Events that are the most interesting are:
- Uplinks
- ACK
- Downlink, scheduled and sent 
- ADR
- JOIN

Agregating or having a combine trace of Join request and Join Accept will make sense, same for ACKs.
MAC commands should be clearly visible when you look for more details, would also be interesting to have a quick glance of those in the small view (like icons or shortened text).

htdvisser, 2019-04-18:
We have the following important events for actual LoRaWAN traffic.

For uplinks:

- `(gs|ns|as).(up|down)(.data|.join|.rejoin)?.receive`
- `ns.up.receive_duplicate`
- `(gs|ns|as).(up|down)(.data|.join|.rejoin)?.forward`
- `(gs|ns|as).(up|down)(.data|.join|.rejoin)?.drop`
- `js.join.accept`
- `js.join.reject` (not sure if we would actually see this event in the console)

For downlinks:

- `as.down.data.receive`
- `as.down.data.forward`
- `as.down.data.drop`
- I couldn't find downlink events in the NS, @rvolosatovs?
- `gs.down.send`
- `gs.down.tx.success`
- `gs.down.tx.fail`

htdvisser, 2019-04-18:
With `gs.up.receive` you'll get a `type.googleapis.com/ttn.lorawan.v3.UplinkMessage` payload. Interesting fields to show are:

- The message type `payload.m_hdr.m_type`
- The device address in `payload.mac_payload.f_hdr.dev_addr` (for data messages) or EUIs in `payload.join_request_payload.{dev,join}_eui`

For `as.up.data.forward` you'll get a `type.googleapis.com/ttn.lorawan.v3.ApplicationUp` payload. Interesting fields to show are:

- The port number `uplink_message.f_port`
- The frame counter `uplink_message.f_cnt`
- The payload `uplink_message.frm_payload` and `uplink_message.decoded_payload` if available.

In both you'll also find the following interesting fields:

- The modulation settings settings `settings.data_rate.lora.{spreading_factor,bandwidth}` (for LoRa) or `settings.data_rate.fsk.bit_rate` (for FSK)
- The frequency `settings.frequency`
- The rx metadata, with specifically the `rssi` and `snr`

rvolosatovs, 2019-04-18:
@htdvisser to answer https://github.com/TheThingsNetwork/lorawan-stack/issues/553#issuecomment-484506643 there are no downlink events defined in NS

kschiffer, 2019-04-19:
Let's also put the current wireframe of the event component here:
[image]
It was discussed briefly [here](https://github.com/TheThingsIndustries/lorawan-stack/issues/1071#issuecomment-446141755). Regard more the interaction principles than the stubbed data here, which is not adapted to the latest state.

I think it makes indeed sense to show event types of different types represented by different "ListEntry" components. The most straightforward way I see fit would be to differentiate between LoRaWAN related events and simpler CRUD(-type) messages from the stack. We could then build up from this differentiation as we go.
Another question to consider is whether we should make a clearer distinction these two, as they might serve different purposes and showing them tied together in one view by default might actually be disadvantageous. We would tackle this by using proper filtering options. Yet, separation into different views might be more in line with the user story (entity logging as opposed to traffic and network status monitoring). Input welcome, @Sypheos maybe?

htdvisser, 2019-04-19:
The console should be able to work with any kind of event (even events it doesn't know), so I think we should start with a "generic event" and make them more specific if we recognize them as specific events:

- Generic Event (shows IDs and event name/description)
  - CRUD Event (adds a nice icon)
    - Update Event (shows the updated fields)
  - Uplink Event (adds a nice icon and some metadata: freq, sf, bw)
    - Uplink Data Event (adds the DevAddr)
      - Application Uplink Data Event (adds the decrypted/decoded payload)
    - Uplink Join Event (different icon, adds the EUIs)
  - Downlink Event (adds a nice icon and some metadata: freq, sf, bw)
    - Downlink Join Accept Event (different icon)
    - Downlink Data Event (adds the DevAddr)

htdvisser, 2019-04-19:
After some discussion on Slack we ended up with the following:

The MVP will show table rows for each event that contain the time, identifiers and event name/description. These rows can be expanded to see the full payload.

Gradually we'll extend those table rows with improvements:

- (colored) icons for events of a certain category:
    - `**.create`, `**.update`, `**.delete` indicate Create/Update/Delete operations
    - `**.receive`, `**.forward`, `**.drop` indicate that a message was received/forwarded/dropped
    - `**.up.**`, `**.down.**` indicate uplink or downlink flow
- Fields extracted form the payloads:
    - "@type":"type.googleapis.com/ttn.lorawan.v3.UplinkMessage"
    - "@type":"type.googleapis.com/ttn.lorawan.v3.ApplicationUp"
    - "@type":"type.googleapis.com/ttn.lorawan.v3.DownlinkMessage"
    - More to follow later

bafonins, 2019-04-23:
The structure then will be:
`"@type":"type.googleapis.com/ttn.lorawan.v3.UplinkMessage"`:
* General Information:
  * Message Type: `payload.m_hdr.m_type` (do we omit this entry if it is empty?)
  * Device Address : `payload.mac_payload.f_hdr.dev_addr`
     or 2 next
  * Join EUI : `payload.join_request_payload.join_eui`
  * Dev EUI : `payload.join_request_payload.dev_eui`
* Modulation Settings 
  * Frequency : `settings.frequency`
  * Bit Rate : `settings.data_rate.fsk.bit_rate`
    or 2 next
  * Spreading Factor : `settings.data_rate.lora.spreading_factor`
  * Bandwidth : `settings.data_rate.lora.bandwidth`
  * `rssi` and `snr`. Can i have an example with those fields being present in the event payload?

`"@type":"type.googleapis.com/ttn.lorawan.v3.DownlinkMessage"`:
* General Information:
  * ?
* rest?

The payload uplink_message.frm_payload and uplink_message.decoded_payload if available.
`"@type":"type.googleapis.com/ttn.lorawan.v3.ApplicationUp"`:
* General Information:
  * Port Number : `uplink_message.f_port`
  * Frame Counter : `uplink_message.f_cnt` (do we omit this entry if it is empty?)
* Payload: (show if available)
  * Raw Payload: `uplink_message.frm_payload` (how big can it get?)
  * Decoded Payload: `uplink_message.decoded_payload` (how big can it get? what is the format?)
* Modulation Settings 
  * Frequency : `settings.frequency`
  * Bit Rate : `settings.data_rate.fsk.bit_rate`
    or 2 next
  * Spreading Factor : `settings.data_rate.lora.spreading_factor`
  * Bandwidth : `settings.data_rate.lora.bandwidth`
  * `rssi` and `snr` see above

bafonins, 2019-04-23:
👆 We can also simply render the `<CodeEditor />` component with json payload for the event extended info (for now).

htdvisser, 2019-04-23:
This is a good start. Let's keep things MVP and add stuff later instead of trying to be 100 complete from the start

- `payload.m_hdr.m_type` shouldn't be empty. I suspect it's omitted when 0 (JoinRequest), but that should/will be fixed.
- Make sure it doesn't crash if neither `mac_payload` nor `join_request_payload` is present (on RejoinRequests or Proprietary message types)
- `rssi` and `snr` aren't "modulation settings" (that's also why they aren't in `settings`)
- `rssi` is a number (usually without decimals and usually between -130 and 0)
- `snr` is a number (usually without decimals and usually between -20 and +20)
- `DownlinkMessage` General Information has `payload.m_hdr.m_type` and may have `payload.mac_payload.f_hdr.dev_addr`
- `DownlinkMessage` has similar "Modulation Settings" as `UplinkMessage`
- Don't omit `uplink_message.f_cnt`. If it's not in there, it's `0`. I'll see if we can fix that (similar problem as the first point)
- `uplink_message.frm_payload` can go up to ~250 bytes
- `uplink_message.decoded_payload` is an object, can get as big as the user makes it

johanstokking, 2019-04-23:
@bafonins note that `ApplicationUp` contains one of these:

```
  oneof up {
    ApplicationUplink uplink_message = 3;
    ApplicationJoinAccept join_accept = 4;
    ApplicationDownlink downlink_ack = 5;
    ApplicationDownlink downlink_nack = 6;
    ApplicationDownlink downlink_sent = 7;
    ApplicationDownlinkFailed downlink_failed = 8;
    ApplicationDownlink downlink_queued = 9;
    //ApplicationInvalidatedDownlinks downlink_queue_invalidated = 10;
    ApplicationLocation location_solved = 11;
  }
```

I commented the one out that won't be send upstream (field 10).

So when you see `ApplicationUp`, you'd have to look in the object to see which one it is. I see that you're referring to `uplink_message.f_port` under "general information", but that's not general at that level.

johanstokking, 2019-04-25:
So we decided offline that we start with the events and expanding the payload.

When we want to nicely show the contents, here's what we want per sub-message:

### General

The end device identifiers, very importantly with the `dev_addr`.

### `uplink_message`

* General Information:
  * Port Number : `uplink_message.f_port`
  * Frame Counter : `uplink_message.f_cnt` (do we omit this entry if it is empty?)
* Payload: (show if available)
  * Raw Payload: `uplink_message.frm_payload` (how big can it get?)
  * Decoded Payload: `uplink_message.decoded_payload` (how big can it get? what is the format?)
* Metadata, see @htdvisser reply above

### `join_accept`

No data, just a nice activation icon. Do note that the `dev_addr` is very relevant here, but I suppose this is shown always.

### `downlink_{queued|sent|ack|nack|failed}`

The payload here is the concerning downlink message;

* General Information:
  * Port Number : `downlink_message.f_port`
  * Frame Counter : `downlink_message.f_cnt` (always >= 1)
  * Confirmed: `downlink_message.confirmed` (bool)
  * Priority: `downlink_message.priority` (always set, always relevant to show, see API. We can have an icon for this maybe?)
* Raw payload (always available, we don't have `decoded_payload` here anymore). Same length limitations apply

In the case of `downlink_failed`, we also have the error. This is the same as any other error payload you get when a call fails. There should be a descriptive error message that is relevant to the user.

### `location_solved`

We don't fire these yet, but see the API `ApplicationLocation`, show `service` and `location` coordinates. We can beautify this by rendering a map later.

closed 2019-04-30 by bafonins