TheThingsNetwork/lorawan-stack #554: Store pending MAC state apart from pending session

rvolosatovs, 2019-04-18:
**Summary:** 

According to 1.1 spec, NS should only assume that device switched session after a join request once an uplink is received using the session context set up in the join-accept. If an uplink received belongs to the old session, the old session context shall be restored. This means that the old MAC state shall be restored as well.

**Why do we need this?**

To be compliant with the spec.
An example, why this is necessary: a malicious actor could use the replay attack in order to hurt the communication of the network with the device, since NS would lose the device's MAC state on each such malicious join request received.
This also makes implementation of #8 easier and consistent with Join-Request handling

**What is already there? What do you see now?**

MAC state being reset on each join request

**What is missing? What do you want to see?**

MAC state being reset when NS knows for sure that the device lost the MAC state

**How do you propose to implement this?**

https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/pending-mac-state I worked on this right before going on holiday, so it's based on the `master` from a month ago

I want to rebase this before proceeding with anything else MAC-related in NS to avoid creating more merge conflicts

**Can you do this yourself and submit a Pull Request?**

yes

rvolosatovs, 2019-04-18:
cc @johanstokking @htdvisser

johanstokking, 2019-04-23:
@rvolosatovs 

1. isn't the problem that we reset the MAC state on join-accept, while we should only reset the MAC state on the first uplink that confirms the `pending_session`?

2. can you clarify the "how do you propose to implement this" seeing that you've spent some time on it now?

rvolosatovs, 2019-04-23:
1. This is a problem and introducing a "pending MAC state" is one of the ways to fix it.
2. Whenever `PendingSession` is set, `PendingMACState` is set as well. (so for now, `PendingMACState` is set only on Join-Request). When scheduling NS uses `PendingSession` and `PendingMACState` if set, otherwise `Session` and `MACState`. When matching NS uses either `PendingSession` and corresponding `PendingMACState` or `Session` and `MACState` depending on the `DevAddr`. In rare cases where `PendingSession` and `Session` have same `DevAddr`, it may use both. `AsNs` RPCs modify `PendingMACState`, if set, otherwise - `MACState`

johanstokking, 2019-04-23:
1. OK, why is this method chosen? What is the downside of not resetting the MAC state on join-accept and only do that on session confirmation by uplink of the pending session?

rvolosatovs, 2019-04-23:
Because this way we only need to construct the MAC state only once.
For scheduling and for uplink matching we need to construct the "pending MAC state"(and apply e.g. CFList from the associated join-accept), however if we store the pending MAC state as a field we can simply use it directly whenever we need it and we can always make sure that it's the correct state.

PR #561 (2019-05-01): Pending MAC state

**Summary:**

Refs #8 
Closes #554 

**Changes:**

- Add and support `PendingMACState`
- Fix some logging statements
- Fix test logger output
- Minimal consistency fixes
- Support downlink in response to Rejoin-request

**Release Notes: _(optional)_**

- Improved OTAA join procedure security
closed 2019-05-01 by rvolosatovs