TheThingsNetwork/lorawan-stack #558: Support Client Auth Check for LNS Gateways

KrishnaIyer, 2019-04-23:
**Summary:** 

With #366 we support various auth modes for the Basic Station on the CUPS/GCR server. 
When Client Auth is enabled on the gateway, the LNS server must verify the credentials (Ex: Auth Token) to authenticate it.

**Why do we need this?**

Verify gateway credentials before connection when Client Auth is enabled.

**What is already there? What do you see now?**

No check for client auth.

**What is missing? What do you want to see?**

Check for client auth and validate it if enabled.

**Environment:**

Gateways with the Basic Station forwarder.

**How do you propose to implement this?**

- For Auth tokens, check if they are sent as part of the header and validate them.
- For Client-side TLS, validate the Client cert.

**Can you do this yourself and submit a Pull Request?**

Yes

htdvisser, 2019-04-23:
We've postponed TLS Client Auth, so let's leave that for later, perhaps together with #137

KrishnaIyer, 2019-04-23:
Fine by me. Moved to a later milestone.

KrishnaIyer, 2020-08-20:
With #2948, we enforce Auth by default. When we support Client TLS, this will remain the case. This issue is no longer relevant.

closed 2020-08-20 by KrishnaIyer