TheThingsNetwork/lorawan-stack #580: Add unique id to events

bafonins, 2019-04-29:
**Summary:** 

We need to uniquely identify events.
References: https://github.com/TheThingsNetwork/lorawan-stack/pull/570

**Why do we need this?**

Currently a combination of time and name is used as the key, however this can cause overlaps.

**What is already there? What do you see now?**

1. The `Event` proto, see https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/events.proto

**What is missing? What do you want to see?**

1. The `id` entry that uniquely identifies the event

**Environment:**

stack

**Can you do this yourself and submit a Pull Request?**

I can do that with guidance from someone who is familiar with the way events work.

kschiffer, 2020-09-10:
As far as I can see, events with the exact same timestamp should not happen, meaning that

*   we can use the timestamp as event key
*   the occasional events with duplicate timestamps need to be addressed in the backend (#3209)
*   we can use logic in the console to check for events with duplicate timestamps and discard those (#3098)
*   we might want to close this issue

cc @johanstokking

johanstokking, 2020-09-10:
My proposal would be:

- Add a new field to `Event` proto
- Fill that when creating the event: https://github.com/TheThingsNetwork/lorawan-stack/blob/3c1ab3a4e8a443e8e28883bc05df27369e110c4a/pkg/events/builder.go#L45-L50
- With an ID that's like correlation ID: `ulid.MustNew(ulid.Now(), rand.Reader).String()`

PR #3211 (2020-09-10): Add unique event identifiers

#### Summary

Closes #580 

#### Changes

- Add unique identifiers to events

#### Testing

Unit testing should cover this.

##### Regressions

No expected regressions.

#### Notes for Reviewers

I chose `UniqueID`/`unique_id` over `ID`/`id` since we already have `Identifiers`/`identifiers`, and I thought it would be confusing to have both (even though indeed they are documented). Naming can still be changed, of course. 

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards c [... cut]
closed 2020-09-10 by adriansmares