TheThingsNetwork/lorawan-stack #584: General Settings view for Devices

kschiffer, 2019-04-29:
**Summary:** 
The console needs to allow users to change general data of a device, similar to how we allow it for applications already.

**Why do we need this?**
It's a basic feature.

**What is already there? What do you see now?**
- SDK support for updating devices: #278 
- The device subview, currently with the overview only #376 
- The "Add Device" view: #573 
  - We should be able to reuse a lot here from the form

**What is missing? What do you want to see?**
- Info on what data can be edited after device creation (cc @htdvisser)
- We don't need wireframes or screen design here
- Implementation

**Environment:**
Console in browser.

**How do you propose to implement this?**
Basically, use the add device form and strip a couple of fields.

**Can you do this yourself and submit a Pull Request?**
Yes.

htdvisser, 2019-04-30:
Question: Will we have a "view single device" screen or is that the same as the "update" screen (possibly with the inputs as read-only)?

For the update screen I think we should start by extending the create screen from #573. The main difference is that you can not change the ID or EUIs.

After this we should decide what additional fields we should add to *both* create and update, and what fields we only want to show in update.

Things that will become useful after users create their first device:

- `formatters` (payload formatters)
- `mac_settings` (all optional)

kschiffer, 2019-04-30:
> Question: Will we have a "view single device" screen or is that the same as the "update" screen (possibly with the inputs as read-only)?

There is a single device overview screen (⟶ [screen design](https://user-images.githubusercontent.com/5710611/55146622-9e333b00-514d-11e9-80cd-3343ffddceb5.png), implemented via #376) which shows the most important info about the device. To update basic (simple key-value) information of the device, we use the "General Settings" tab. More complex data, such as payload formatters or (later on) device location will be set in dedicated views.

> For the update screen I think we should start by extending the create screen from #573. The main difference is that you can not change the ID or EUIs.

👌

PR #624 (2019-05-06): General Settings view for Devices
#### Summary
Closes #584 

[image]

#### Changes
- Refactor device data form into a dedicated container
- Add general settings view for devices
- Add option to use `<Tab />`'s as nav elements
- Add submit logic for updating devices
- Add disabled styling to checkboxes and radio buttons
- A couple of minor styling tweaks

#### Notes for Reviewers
I have added the link prop to the `<Tabs />` which will render the tab element as `<NavLink />`, as such also handling active state by itself. I decided to not make it a dedicated component, as the functionality could  [... cut]
closed 2019-05-06 by kschiffer