TheThingsNetwork/lorawan-stack #586: The ChannelMask field in the LinkADRReq command mismatch with the frequency_plan

twowinter, 2019-04-30:
Hi,

I am DongDong, a backend engineer from Tencent. We really appreciate TTN's contribution for LoRa community.
Our team is using the CN470 region parameter to test the Stack. We have encountered a issue with the ChannelMask field in the LinkADRReq command mismatch with the frequency plan. Now it has finally been fixed, so I submitted the issue.

**Summary:** 

The ChannelMask field in the LinkADRReq command mismatch with the frequency_plan. This issue exists in the band which have more than 16 channels: US902-928、AU915-928、CN470-510. 

**Steps to Reproduce:** 

1. EndDevice takes the frequency_plan of US902-928 or AU915-928 or CN470-510, which have more than 16 channels.
2. EndDevice joins, and send any data after joined.
3. NetworkServer would send the mismatched LinkADRReq command.

**What do you see now?**

When end-device takes the CN_470_510_FSB_11, it get the LinkADRReq: {{03 00 FF 00 01},{03 00 FF 00 11},{03 00 FF 00 21},{03 00 FF 00 31},{03 00 FF 00 41},{03 00 FF 00 51}}
When end-device takes the CN_470_510_FSB_1,  it get the LinkADRReq: {{03 00 00 00 01},{03 00 00 00 11},{03 00 00 00 21},{03 00 00 00 31},{03 00 00 00 41},{03 00 00 00 51}}

This issue is basically the same as another issue(https://github.com/TheThingsNetwork/lorawan-stack/issues/544). I updated the latest code and used the latest frequency plan, but the issue still exists.

**What do you want to see instead?**

When end-device takes the CN_470_510_FSB_11, it should get the LinkADRReq: {{03 00 00 00 01},{03 00 00 00 11},{03 00 00 00 21},{03 00 00 00 31},{03 00 00 00 41},{03 00 FF 00 51}}
When end-device takes the CN_470_510_FSB_1,  it should get the LinkADRReq: {{03 00 FF 00 01},{03 00 00 00 11},{03 00 00 00 21},{03 00 00 00 31},{03 00 00 00 41},{03 00 00 00 51}}

**Environment:**

This issue can be reproduced by any environment.

**How do you propose to implement this?**

I have created a minimized program, [chmask_issue.go](https://github.com/TheThingsNetwork/lorawan-stack/files/3129858/chmask_issue.txt), attatched in this post. The key code is this:

```go
for _, m := range desiredMasks {
	pld := &MACCommand_LinkADRReq {
		ChannelMask:        m.Mask[:],
	}
	fmt.Printf("p = %p %p\r\n", pld.ChannelMask, &m.Mask[0])
}
```

When we pass value from m.Mask array to ChannelMask slice, we know that it actually pass the pointer of the array to the slice. Problem happens here, m.Mask could only change its value in each iteration of the loop, its pointer would be consistent, which is allocated once at the beginning of the loop.

So, each pld.ChannelMask would be overwrited with the m.Mask in the last iteration of the loop:
i. When end-device takes the CN_470_510_FSB_11, all ChannelMask would be overwrited with the m.Mask in the last iteration of the loop, the value is {true, true, true, true, true, true, true, true, false, false, false, false, false, false, false, false}.
ii. When end-device takes the CN_470_510_FSB_1, all ChannelMask would be overwrited with the m.Mask in the last iteration of the loop, the value is {false, false, false, false, false, false, false, false, false, false, false, false, false, false, false, false}.

**Can you do this yourself and submit a Pull Request?**

I would like to recommend 2 solutions.

Solution 1. Passing real memory addresses by array subscript

```go
for i, m := range desiredMasks {
	pld := &ttnpb.MACCommand_LinkADRReq{
		ChannelMask:        desiredMasks[i].Mask[:],
	}
}
```

This solution takes the least amount of code changes. I have made test, this issue is finally fixed.

Solution 2. Passing value by changing the type of ChannelMask to array

```go
type MACCommand_LinkADRReq struct {
	ChannelMask          [16]bool        `protobuf:"varint,3,rep,packed,name=channel_mask,json=channelMask,proto3" json:"channel_mask,omitempty"`
}
```

This solution needs to adjust at least 4 places, but it is the better way to operate in Range loop, no need to care about memory leak.

Which one do you prefer for these two solution? or any other suggestion? I'd like to submit a Pull Request.

htdvisser, 2019-04-30:
Hi @twowinter, thanks for the very detailed issue report!

I'm assigning @rvolosatovs to coordinate this with you and to help you with fixing this if needed!

rvolosatovs, 2019-04-30:
Hello DongDong,

Thanks for submitting this!
Good catch!
I'd prefer the first solution too.

twowinter, 2019-04-30:
Hi @htdvisser  @rvolosatovs ,

Here is the pull request,  feel free to leave a message to me, if you have any question about it.

PR #597 (2019-04-30): Fix ChannelMask in LinkADRReq mismatch with the frequency_plan

#### Summary

Closes #586 

#### Changes

- Pass the memory address of origin array to ChannelMask slice.
closed 2019-04-30 by rvolosatovs