TheThingsNetwork/lorawan-stack #598: Document Message Processors

htdvisser, 2019-05-01:
#### Summary

We need to document the use of message processors.

#### Why do we need this?

This is a frequently used feature (at least in v2), but we haven't documented anything about it in v3.

#### What is already there? What do you see now?

- Processors for CayenneLPP, JavaScript (and gRPC?) are implemented in the AS.
- They can be set with the CLI
    - per device (see `ttn-lw-cli end-devices update -h | grep formatters`)
    - per application (see `ttn-lw-cli applications link set -h | grep formatters`)

#### What is missing? What do you want to see?

- [ ] Documentation on how to configure CayenneLPP formatters
- [ ] Documentation on how to configure JavaScript formatters
    - [ ] Expected function signatures
    - [ ] Environment
- [ ] Documentation on how to implement and configure gRPC formatters

There is also no information on the EndDevice Repository formatters, I don't think that's even implemented yet (@johanstokking?)

#### Can you do this yourself and submit a Pull Request?

Probably best if @johanstokking does this.

johanstokking, 2019-05-01:
Refs #263

htdvisser, 2019-06-25:
Blocked on #401

benolayinka, 2020-06-15:
Superseded by #2713

closed 2020-06-15 by benolayinka