TheThingsNetwork/lorawan-stack #601: LoRaWAN Encoding/Decoding in the CLI

htdvisser, 2019-05-01:
#### Summary

It would be nice to be able to decode/encode LoRaWAN packets directly in the CLI.

#### Why do we need this?

Mostly for debugging purposes.

#### What is already there? What do you see now?

Most of the code needed to do this is already in the CLI for the simulation commands.

#### What is missing? What do you want to see?

Would be cool if you could pipe raw (binary/base64/hex) packets to the CLI, and print them as JSON.

#### How do you propose to implement this?

Create a `decode` command that does something similar to the `processDownlink` func in `simulate.go`, maybe we can even make the decoder call `processDownlink` for downlink messages.

#### Can you do this yourself and submit a Pull Request?

Yes, but no high priority.

htdvisser, 2020-07-14:
@rvolosatovs I guess you overlooked this when you implemented https://github.com/rvolosatovs/lorawan-codec. Would be nice if you could add this to our own CLI instead of having a separate tool for it.

rvolosatovs, 2020-07-14:
I think it's out-of-scope of the CLI, but I will add it.

neoaggelos, 2021-08-01:
Quite a lot of initial work for LoRaWAN decoding in https://github.com/TheThingsNetwork/lorawan-stack/tree/cli-decode-lorawan

htdvisser, 2021-08-04:
Since @neoaggelos isn't going to finish this before the v3.14.1 deadline, I'll move it to v3.14.2.

neoaggelos, 2021-08-17:
LoRaWAN encoding is already implemented and present in the `ttn-lw-cli simulate join-request --dry-run` family of commands (also for uplink and downlink messages), so focusing on decoding only.

PR #4544 (2021-08-20): LoRaWAN decoding CLI command

#### Summary

Closes #601 

#### Changes

- Support `base64` and `hex` input formats (only for `[]byte` inputs)
- Add `ttn-lw-cli decode lorawan` command, which accepts LoRaWAN frames (as hex or base64 formatted strings), decodes them and prints them in JSON format.

#### Testing

Test with captured LoRaWAN traffic. See examples.

##### Regressions

None

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, accordi [... cut]
closed 2021-08-20 by neoaggelos