TheThingsNetwork/lorawan-stack #629: Console MVP

kschiffer, 2019-05-03:
#### Summary
This is an umbrella issue to track the status of the Console MVP.

#### Status
- [ ] Applications
  - [x] Application List
  - [x] Add Applications
  - [x] Overview
  - [ ] End Device List ⟶ issue following
  - [ ] End Devices ⟶ #25 
    - [x] Add End Devices ⟶ #573 
    - [x] Overview ⟶ #376 
    - [ ] Payload Formats ⟶ #71 (issue for devices needed)
    - [ ] End Device Event/Data Stream ⟶ issue following
    - [x] General Settings (Edit End Devices) ⟶ #624 
  - [x] Application Event/Data Stream ⟶ #522 
  - [x] Linking ⟶ #214 
  - [ ] Payload Formats ⟶ #71 (issue for apps needed)
  - [x] API Keys ⟶ #142 
  - [x] Collaborators ⟶ #449 
  - [x] General Settings (Edit Applications)
- [ ] Gateways ⟶ #26 
  - [ ] Gateway List (only stubbed atm, needs issue)
  - [ ] Overview ⟶ #243 
  - [ ] Add Gateways ⟶ #628 
  - [ ] Gateway Event/Data Stream ⟶ needs issue
  - [ ] General Settings (Edit Gateways) ⟶ needs issue
  - [ ] Collaborators ⟶ needs issue
- [ ] SDK ⟶ #30 
  - [ ] Entity CRUD
    - [x] Applications
    - [x] End Devices ⟶ #278 
    - [x] Gateways ⟶ #620 
  - [x] Stack Config ⟶ #588 
    - [x] Frequency Plans ⟶ #588 
  - [ ] Event Streams
    - [x] Applications ⟶ #522 
    - [ ] End Devices ⟶ needs issue
    - [ ] Gateways ⟶ needs issue

- [ ] Overall
  - [ ] Internal testing and debugging
  - [ ] Proper error handling ⟶ #41 
  - [ ] Streamline page titles ⟶ #619 
  - [ ] Styling finish
  - [ ] Decision on general overview page
  - [x] User Login (not exactly console)
  - [x] User Registration (not exactly console)
  - [x] Authorization screen (not exactly console)

Undecided whether in MVP:
- [ ] Profile Settings of the user
- [ ] Organizations management
- [ ] First release of SDK

@bafonins @htdvisser @johanstokking please check and add whatever you find missing. I will try to keep this up to date, any help welcome.

johanstokking, 2019-06-28:
Can we add some basic overview to the scope of MVP?

kschiffer, 2019-07-03:
Yes. Updated the description.

kschiffer, 2019-08-05:
I think we can consider this done now since the console has been "released" with 3.1.0. Remaining issues are taken care of.

closed 2019-08-05 by kschiffer