TheThingsNetwork/lorawan-stack #631: Fail absolute timed downlink that's too late to schedule

johanstokking, 2019-05-03:
#### Summary

NS should fail absolute timed downlink that's too late to schedule.

#### Why do we need this?

So that a queued downlink item that can never be scheduled anymore by a GS does not keep halting the downlink flow until the queue is reset.

#### What is already there? What do you see now?

Currently, when you schedule a downlink message with an absolute time in the past or too soon in the future for any GS to schedule, the downlink task fails. However, the downlink item stays first in the queue, so it gets retried and keep failing until the queue is replaced with an item that can be scheduled.

GS now returns an `Aborted` error with error details per downlink path and Rx window that was specified. The error details code can be:

- `ResourceExhausted`: no more resources available, i.e. scheduling conflict
- `FailedPrecondition`: this is not going to work, i.e. too late – this is what should cause failure
- `Unavailable`: try again later, i.e. no clock sync; NS may retry later on this downlink path when there is clock sync (i.e. once there's uplink)
- `Aborted`: try another downlink path, i.e. this gateway doesn't have absolute time

#### What is missing? What do you want to see?

Based on errors returned by GS, NS should decide to fail the downlink and inform AS about that. Basically, when NS sees `FailedPrecondition` on all scheduling attempts, it can fail the downlink.

#### How do you propose to implement this?

NS should decide whether to fail the downlink based on GS error details codes. Suggesting the following;

- If all attempts are either `ResourceExhausted`, `FailedPrecondition`, `Aborted`, fail the downlink
- If any attempt is something else (i.e. `Unavailable`), try again later, like now

#### Can you do this yourself and submit a Pull Request?

If the proposed solution is acceptable for @rvolosatovs, then GS doesn't need a change. I'm open for suggestions that may involve specific GS errors.

rvolosatovs, 2019-05-06:
Sounds good to me!
This only concerns the "absolute time" downlink, right?
What about other cases?
Should NS *only* retry downlink if at least one of the errors in `Unavailable`?

johanstokking, 2019-05-06:
The proposed solution is for any timed downlink so: class A and class B/C with absolute time.

And indeed, it should only retry if there's any `Unavailable`.

Do we want to fail the join-accept too?

johanstokking, 2019-06-20:
There seems to have been regression causing an endless scheduling loop;

```
 DEBUG Process downlink task                    device_uid=test-app.test-dev-2 namespace=networkserver start_at=2019-06-20 21:22:23.445219 +0000 UTC started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Add MAC command to buffer                cid=CID_DEV_STATUS device_uid=test-app.test-dev-2 mac_version=1.0.3 namespace=networkserver phy_version=1.0.3-a started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Use confirmed downlink to get immediate answer cid=CID_DEV_STATUS device_uid=test-app.test-dev-2 mac_version=1.0.3 namespace=networkserver phy_version=1.0.3-a started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Add MAC command to buffer                cid=CID_LINK_ADR device_uid=test-app.test-dev-2 mac_version=1.0.3 namespace=networkserver phy_version=1.0.3-a started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Generated downlink                       ack=false adr=true device_uid=test-app.test-dev-2 f_cnt=2 f_pending=false f_port=0 m_type=CONFIRMED_DOWN mac_count=2 mac_version=1.0.3 namespace=networkserver payload_length=19 phy_version=1.0.3-a priority=HIGHEST started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Schedule downlink                        attempt_rx1=false attempt_rx2=true device_class=CLASS_C device_uid=test-app.test-dev-2 downlink_class=CLASS_C downlink_priority=HIGHEST downlink_type=data namespace=networkserver path_count=1 rx2_data_rate=0 rx2_frequency=869525000 started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Attempt to schedule downlink on gateway  class=CLASS_C gateway_eui=647FDAFFFE007926 gateway_uid=home-tektelic namespace=gatewayserver/io/udp
 DEBUG Attempt to schedule downlink in receive window class=CLASS_C data_rate_index=0 frequency=869525000 gateway_eui=647FDAFFFE007926 gateway_uid=home-tektelic namespace=gatewayserver/io/udp rx_window=2
 DEBUG Failed to schedule downlink in Rx window class=CLASS_C data_rate_index=0 error=error:pkg/gatewayserver/scheduling:no_clock_sync (no clock sync) frequency=869525000 gateway_eui=647FDAFFFE007926 gateway_uid=home-tektelic namespace=gatewayserver/io/udp rx_window=2
 DEBUG Failed to schedule on path               error=error:pkg/gatewayserver/io:tx_schedule (failed to schedule) gateway_uid=home-tektelic grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=gatewayserver request_id=01DDVC9S4C23K2FGFE43NHCK6G
  WARN Finished unary call                      duration=504.859µs error=error:pkg/gatewayserver:schedule (failed to schedule) error_correlation_id=2df7b93da2f64bca8c3166857c0484a9 error_name=schedule error_namespace=pkg/gatewayserver grpc_code=Aborted grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=grpc request_id=01DDVC9S4C23K2FGFE43NHCK6G
  WARN All Gateway Servers failed to schedule downlink attempt_rx1=false attempt_rx2=true device_class=CLASS_C device_uid=test-app.test-dev-2 downlink_class=CLASS_C downlink_priority=HIGHEST downlink_type=data error_0=rpc error: code = Aborted desc = error:pkg/gatewayserver:schedule (failed to schedule) namespace=networkserver rx2_data_rate=0 rx2_frequency=869525000 started_at=2019-06-20 21:22:23.496797 +0000 UTC
 DEBUG Add downlink task after downlink slot    device_class=CLASS_C device_uid=test-app.test-dev-2 namespace=networkserver start_at=2019-06-20 21:22:24.50088 +0000 UTC started_at=2019-06-20 21:22:23.496797 +0000 UTC
```

This repeats every second.

rvolosatovs, 2019-06-21:
That is expected behavior, since the NS does not check the error returned by GS(this issue), it keeps retrying every second https://github.com/TheThingsNetwork/lorawan-stack/blob/6eb8fe05751cd81b0dd16a97a05de67356d830d2/pkg/networkserver/downlink.go#L726

rvolosatovs, 2019-07-03:
@johanstokking now that #882 is merged is this still high priority and/or blocking release?
The only thing missing here is notifying AS when `ClassBC` paths are set and not applicable and in very rare cases when `ClassBC` absolute time is expired.

johanstokking, 2019-07-04:
Go through the error details of the error you get from `nsgs.ScheduleDownlink`.

## Class A downlink

Try Rx1 and Rx2 of the uplink tokens that you have.

If all attempts fail, retry on the next uplink.

## `ClassBC` set

Only try in class B/C downlink.

### Fixed downlink paths

If all attempts say `NotFound` (gateway not connected) or `Corruption` (bad uplink token) or `FailedPrecondition` (something else); fail and drop the downlink, inform AS

We _may_ in the future add limited retry logic in the NS because the gateway may connect in the near future.

### Absolute time

If all attempts say `Aborted`, `ResourceExhausted` or `FailedPrecondition`, fail and drop the downlink, inform AS.

### Other class C downlink

If no fixed downlink paths and/or no absolute time, the NS can retry with new uplink tokens it receives.

rvolosatovs, 2019-07-05:
Blocking #936

rvolosatovs, 2019-07-10:
Blocked by #976

rvolosatovs, 2019-07-10:
Blocked by #967

PR #952 (2019-07-30): Fail downlink based on GS errors, fix retry logic

#### Summary

Closes #631
Closes #944 

#### Changes

- Only retry Class C downlink slots, which can be retried
- Fail downlinks based on GS errors
- Fix unscheduled valid application downlink being removed from the queue, if it is updated
closed 2019-07-30 by htdvisser