TheThingsNetwork/lorawan-stack #637: The CLI does not set the gateway server address of new gateways

htdvisser, 2019-05-06:
#### Summary

Just like we do with the network/application/join server address end devices, we need to register the gateway server address for a newly created, and make sure we don't get connection stats

#### Steps to Reproduce

1. Create a gateway
2. Get the gateway server address of the gateway

#### What do you see now?

It's empty

#### What do you want to see instead?

It should be the one that was configured in the CLI, just as we do with end devices.

#### How do you propose to implement this?

- Rename [`compareServerAddresses`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/end_devices.go#L745) so that it contains EndDevice and create a similar function for gateways.
- This function should be called before [calling the gateway server](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/gateways.go#L341) for getting the connection stats
- On gateway create, set the gateway server address [just as we do for end devices](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/end_devices.go#L312)

#### Can you do this yourself and submit a Pull Request?

Sure, but I believe @furtiman was looking for a nice LoRaWAN Stack (or Go in general) issue.

furtiman, 2019-05-06:
👀 👍

furtiman, 2019-05-14:
@htdvisser I do get the one I configured back though

I create it:
```
ttn-lw-cli gateways create gtw2 \
  --user-id admin \
  --frequency-plan-id EU_863_870 \
  --gateway-eui 00800000A00009EA \
  --enforce-duty-cycle \
  --gateway-server-address 01234567
 DEBUG Using access token (valid until 11:16PM)
 DEBUG ccResolverWrapper: sending update to cc: {[{localhost:8884 0  <nil>}] }
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc00062d220, CONNECTING
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc00062d220, READY
 DEBUG Finished unary call                      duration=173.828049ms grpc_method=Create grpc_service=ttn.lorawan.v3.GatewayRegistry namespace=grpc
{
  "ids": {
    "gateway_id": "gtw2",
    "eui": "00800000A00009EA"
  },
  "created_at": "2019-05-14T20:30:44.089Z",
  "updated_at": "2019-05-14T20:30:44.089Z",
  "version_ids": {

  },
  "gateway_server_address": "01234567",
  "frequency_plan_id": "EU_863_870",
  "antennas": [
    {
      "location": {
        "source": "SOURCE_REGISTRY"
      }
    }
  ],
  "enforce_duty_cycle": true
}
```

And then get the info with the gateway-server flag, where I can see the one I have assigned while creating the gateway:

```
ttn-lw-cli gateways get "gtw2" --gateway-server-address
 DEBUG Using access token (valid until 11:16PM)
 DEBUG ccResolverWrapper: sending update to cc: {[{localhost:8884 0  <nil>}] }
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc0005deb90, CONNECTING
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc0005deb90, READY
 DEBUG Finished unary call                      duration=48.748095ms grpc_method=Get grpc_service=ttn.lorawan.v3.GatewayRegistry namespace=grpc
{
  "ids": {
    "gateway_id": "gtw2",
    "eui": "00800000A00009EA"
  },
  "created_at": "2019-05-14T20:30:44.089Z",
  "updated_at": "2019-05-14T20:30:44.089Z",
  "version_ids": {

  },
  "gateway_server_address": "01234567"
}
```

Do I get something wrong?

johanstokking, 2019-05-15:
@furtiman the idea is that you don't have to explicitly specify it, just like with end devices.

Also, there should be an address check before getting the connection stats.

furtiman, 2019-05-15:
So going back here, whenever I want to get the info of the ned device, it also does not get me the server addresses, unless i explicitly specify it:

Without specifying:
```go run ./cmd/ttn-lw-cli end-devices get app1 dev1
grpc_service=ttn.lorawan.v3.EndDeviceRegistry namespace=grpc
{
  "ids": {
    "device_id": "dev1",
    "application_ids": {
      "application_id": "app1"
    },
    "dev_eui": "0004A30B001C0530",
    "join_eui": "800000000000000C"
  },
  "created_at": "2019-05-14T20:49:23.160Z",
  "updated_at": "2019-05-14T20:49:23.160Z"
}
```

With `--network-server-address` flag:
```
go run ./cmd/ttn-lw-cli end-devices get app1 dev1 --network-server-address
grpc_service=ttn.lorawan.v3.EndDeviceRegistry namespace=grpc
  WARN Registered Network Server address does not match CLI configuration configured=localhost registered=1234567
grpc_service=ttn.lorawan.v3.JsEndDeviceRegistry namespace=grpc
{
  "ids": {
    "device_id": "dev1",
    "application_ids": {
      "application_id": "app1"
    },
    "dev_eui": "0004A30B001C0530",
    "join_eui": "800000000000000C"
  },
  "created_at": "2019-05-14T20:49:23.160Z",
  "updated_at": "2019-05-14T20:49:23.160Z",
  "network_server_address": "1234567",
  "join_server_address": "localhost"
}
```

I'm sorry for a lot of questions, but I want to make sure I understand the issue correctly. 

I understand that we want to display the server gateway server address for the gateways, but I don't see it working for the end-devices that are referred to as an example. Am I doing something wrong?

PR #697 (2019-05-21): CLI gateway server address default and mismatch check

#### Summary

Closes #637 

#### Changes

- Added default (configured for CLI) gateway server address assignation to a gateway created with `gateways create` without specified `--gateway-server-adress`
- Added gateway server address mismatch check in `connection-stats` command

#### Notes for Reviewers

As requested by @htdvisser . The same procedures happen to end-devices' `ns`, `js`, `as`, so there is no reason gateways should not have it for `gs`.

Creating gateway without `--gateway-server-address`, `localhost` set as default:
```
go r [... cut]
closed 2019-05-21 by furtiman