TheThingsNetwork/lorawan-stack #654: Default value handling in console / sdk

kschiffer, 2019-05-07:
#### Summary
Following up the discussion starting here: https://github.com/TheThingsNetwork/lorawan-stack/pull/639#discussion_r281178795, we need to discuss handling default values in the client (console / sdk)

#### Why do we need this?
Consistency; better User and Developer Experience

#### What is already there? What do you see now?
- Default values handled by the backend, see eg here: #47 
- Default values being set in the CLI, eg: https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/end_devices.go#L310-L318
- Default values being set by the JS SDK: https://github.com/TheThingsNetwork/lorawan-stack/blob/master/sdk/js/src/service/devices/index.js#L193-L200
- Default values being set by the console, to be displayed in fields: https://github.com/TheThingsNetwork/lorawan-stack/pull/639
- Generated values being communicated with empty fields and placeholders:
[image]

#### What is missing? What do you want to see?
- A clear strategy on handling default values from various sources
- A clear strategy on communicating default values in the console

#### Environment
Various.

#### How do you propose to implement this?
Subject of discussion.

Copy of [my comment in the PR](https://github.com/TheThingsNetwork/lorawan-stack/pull/639#discussion_r281523080):
> This is about whether to set default values in the (JS) SDK and/or in the console itself (like server addresses here, which are not set by the backend, but we do have an idea of what a default should be in a specific context). Whereas I think we should do both, because (non console) SDK users will likely want to make use of that functionality as well. In the console we can determine these default values too and use them to prefill form values, to improve clarity.

> A special case would be randomly generated key values which can be generated either by the client (SDK / console) or also by the backend (to ensure validity in eg. DevAddr). Here, in any case, I think showing the value up front to the user does not add any value and can be confusing rather.

#### Can you do this yourself and submit a Pull Request?
Is a shared responsibility.

cc @htdvisser

kschiffer, 2019-05-07:
Copying [comment](https://github.com/TheThingsNetwork/lorawan-stack/pull/639#discussion_r281543345) of @htdvisser:

I think we should avoid adding too much magic to the SDKs. I think defaults should either be the responsibility of the server, in which case the client doesn't know anything about them and doesn't send them to the SDK (and therefore backend) on create. The console for instance can just show a placeholder "use server default". Other defaults can be the responsibility of the client, in which case the default is shown as a placeholder, and the value is sent explicitly to the SDK on create. I think the SDKs should do exactly (and only) what the clients tell them to do, and not inject magic values.

For addresses there are multiple use cases:

OTAA and bring your own JoinEUI and DevEUI
OTAA and bring your own DevEUI and request cluster JoinEUI
ABP and bring your own DevAddr
ABP and request cluster DevAddr
For keys there are only two:

bring your own keys
securely generate keys (best to do that client side)

kschiffer, 2019-05-07:
> I think we should avoid adding too much magic to the SDKs

With all component split functionality taken care of by the SDK, there already is naturally quite some magic happening in the SDK. Adding convenience as opposed to doing complex querying logic is one of the reasons why we are building the SDK.

> I think the SDKs should do exactly (and only) what the clients tell them to do, and not inject magic values.

Agreed. This is why adding default values in the SDK is currently triggered by a [config option `setDefaults`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/sdk/js/src/service/devices/index.js#L215) (currently defaults to `true`, but that could be changed), [similarly to how the CLI deals with this](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/ttn-lw-cli/commands/end_devices.go#L310). This way, we leave the choice to the user, whether he wants default value magic happening or not, it also helps us separating responsibilities in the console. Do you think this is a way to go?

johanstokking, 2019-05-07:
>With all component split functionality taken care of by the SDK, there already is naturally quite some magic happening in the SDK

That's definitely different magic than the default values. That there's device field splitting, which is strictly functional, is not a reason to add default values because there's magic already. I think we agree on this, but just to be sure.

Regarding the default values that are set by the JS SDK and CLI, yes, those _cannot_ be set by the server. The Identity Server does not know the which cluster the client is using. That is by definition local configuration.

There is no point in comparing this to device addresses, EUIs and keys, because that is not cluster related.

kschiffer, 2019-05-07:
> That's definitely different magic than the default values. That there's device field splitting, which is strictly functional, is not a reason to add default values because there's magic already. I think we agree on this, but just to be sure.

👌

> Regarding the default values that are set by the JS SDK and CLI, yes, those cannot be set by the server. The Identity Server does not know the which cluster the client is using. That is by definition local configuration.

Question remains: is it ok to **optionally** allow setting local configuration defaults in the SDK or not? I'm really talking only about the ones that we would fill with default values in the console as well. The same way we are currently doing it in the CLI.

> There is no point in comparing this to device addresses, EUIs and keys, because that is not cluster related.

But these values are currently also generated locally (cli / sdk)? 
Same thing: should the SDK optionally be allowed to take care of generating necessary values or should all of this be handled strictly outside of it. In our case meaning that we would do it in the console instead.

johanstokking, 2019-05-07:
Okay I see the confusion arising here.

We do allow overriding addresses that are filled out with the client’s configuration. However, in case of devices, if an address is not a cluster address, there is no call made. We may change this later. 

An example is a cluster with only a JS and Console. The user must specify a remote NS address, but there is no call to that NS.

With the addresses and EUIs now I understand the confusion. These values come, indeed, from the client. However, the default `DevAddr` and `JoinEUI` are given by the server. So there should be rpcs that the client (Console, CLI) calls and uses as default values. They may be overridden by the user.

As for key generation, that’s the only thing left after addresses, `DevAddr` and `JoinEUI` (right?), I think that should be done by the SDK.

htdvisser, 2019-05-09:
I'm not very happy with the idea of hiding behavior from the user. The user needs to see what's happening, so that they can decide if it's indeed the behavior they want. The SDK shouldn't make assumptions on how a user intends to use the system.

Maybe we shouldn't use the term "default", at least not for the following 3 situations:

1. The value is optional
  - input is shown as empty
  - field is omitted from request (not present in field mask or payload)
  - maybe with a placeholder "optional"
  - maybe with a note saying what the consequence of an empty field is (there may be a default value, or functionality may be disabled completely)
  - for example: omitting the `network_server_address` in ApplicationLink, not configuring the EndDevice's `factory_preset_frequencies`.
2. The value can be assigned (by backend):
  - input is initially empty
  - input can be filled by user
  - there is a button to populate it with a value requested from the backend
  - for example: `dev_addr`, `join_eui`
3. The value can be randomly generated:
  - input is initially empty
  - input can be filled by user
  - there is a button to populate it with a random value
  - for example: root keys, session keys

In the console, 2 and 3 can just be a single component (an input with a "populate" button), but the two have a different "populate function".

This covers @kschiffer's bullet 1, 3, 4, 5.

The fourth situation (@kschiffer's bullet 2) is that the value is suggested by the client, based on its configuration:
  - input is pre-filled
  - input can still be changed by user
  - for example: gateway/network/application/join server addressses of gateway/end device

This fourth case can also be extended to "... based on previous user behavior" such as pre-filling the same LoRaWAN MAC/PHY version that was used in the previous CREATE.

kschiffer, 2019-05-09:
Thanks for the clear write-up. A couple of remarks:

> 1. The value is optional
[…]
> - maybe with a placeholder "optional"

We do that currently in reverse, by marking and enforcing required fields, which I think is the better option, because it is common.

> - there is a button to populate it with a value requested from the backend
[and]
> - there is a button to populate it with a random value

I think for this, we could then rather do it as we do in the V2 console. The input has a switch to toggle between "generated" and "self-entered" mode. The advantage is that we can decide what mode to set by default based on what we assume the user will most likely want to do, thus sparing the user a click in common use cases. 
In this case, we would not show the generated value to the user upfront (but obviously a note stating that a value will be generated), but since it's basically random numbers, I don't know if she would gain anything from it.

With regards to the SDK, I understand now that we do not want to set client suggested values (your fourth scenario) in the SDK at all, also not with a config flag. This also holds true for not generating random values (your point 3) in the SDK.

johanstokking, 2019-05-09:
>In this case, we would not show the generated value to the user upfront (but obviously a note stating that a value will be generated), but since it's basically random numbers, I don't know if she would gain anything from it.

Note that we have to classes here:
1. Random values (keys)
2. Server provided required fields, i.e. ABP `DevAddr` and OTAA `JoinEUI`

I think that for especially class 2, it is very valuable to show them to the user upfront. You need to make those calls anyway before hitting the server as those fields are required.

kschiffer, 2019-05-09:
Understood. The only problem I still see then is
> (eg. avoiding having to click "Generate" five times when creating ABP devices)

This is dependent on whether it is common to generate most of the values or not.

Possible solution:
- Use an input with "Request Value" button for backend requested values
- Use a toggle input (from my previous comment) for client generated values

I confess it is getting a little _goldplating_, but still worth considering. For MVP we could stick with one field type for both.

johanstokking, 2019-05-09:
For the user, generating `DevAddr` or keys is the same thing. Technically, the former is done by the server, the latter by the client.

Hence @htdvisser suggestion for a populate button with some function that either does something locally or calls an rpc.

kschiffer, 2019-05-09:
I understand that. But consider the necessary interaction if a user wants to create an ABP Device in MAC 1.1.0+, using only generated values. She would have to click a generate button five times for `dev_addr`, `f_nwk_s_int_key`, `s_nwk_s_int_key`, `nwk_s_enc_key` and `app_s_key`,  – as opposed to no necessary further interaction if we have a toggle input.

Question: is this user story (and comparable ones) common enough to be concerned about this UX issue?

johanstokking, 2019-05-09:
I see your point; fine to make the necessary population on create in this case.

kschiffer, 2019-05-14:
We can close this for now.

closed 2019-05-14 by kschiffer