TheThingsNetwork/lorawan-stack #659: Use error library to streamline error handling in frontend

kschiffer, 2019-05-07:
#### Summary
In order to progress with proper display of erros in the frontend via #41, we need to first implement a proper way to handle these errors.

#### Why do we need this?
We don't reliably handle errors and exceptions in the console yet.

#### What is already there? What do you see now?
Occasional error checks without proper message display for the user, like eg. [here](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/console/views/application/index.js#L140).

#### What is missing? What do you want to see?
A reliable, streamlined and agreed-upon way handle errors in the frontend.

#### Environment
Webui in browser.

#### How do you propose to implement this?
First of all, we need to consider the types of errors that can occur in the frontend:
1. API returned backend errors with proper error object
2. API returned backend errors without error object or with an unusual error object
3. Handled Frontend Errors (eg. as a result of a handled backend error, or try/catch blocks)
4. Unhandled Frontend Exceptions

Our intentions by error case:
1. We usually want to preserve the error information to be able to display relevant information to the user
2. We usually want to be able to inform the user that the error originated from the stack
3. We want to communicate helpful error information to the user, at best hinting towards issue resolution
4. We want to fallback to a generic error message

It should be noted that:
- most errors we currently have in the console originate from errors in API calls
- handling errors should encompass proper i18n of error messages in all cases

**Proposed solution:**
Add an error lib, to create custom error classes extending the `Error()` class which will already help us getting useful stack trace information:
- `ConsoleError()`
  - overall error superclass for the console, to be inherited
  - Props:
    - Useful debug info (to be passed to sentry also later on)
    - Other misc. optional metadata (eg. troubleshooting info for the user)
  - Methods:
    - Helper functions to retrieve general info about the error (eg. sth like `isBackend`, `isFatal`, `isNotFound`)
- `BackendError()`
  - is instantiated with the error object returned from the backend
  - Message provided ids can be used to display translated messages (via `webui/locales/.backend`)
  - Can fallback to a generic error message for backend errors
- `FrontendError()`
  - Is instatiated with extended information, defined in place, including:
    - Error ID
    - Error Title (as intl.Message)
    - Error Message (as intl.Message)

The goal here is to end up with a reliable error object structure that we can use to safely extract necessary info from.

By using this strategy, the errors could be instantiated and thrown from anywhere in the react tree to be caught by error boundaries which will pass the error to an `<Error>` component, handling proper display based on the context. In case of plain `Error`'s from unhandled exceptions the component could display a fallback message.

#### Can you do this yourself and submit a Pull Request?
Yes. But I need @bafonins approval for this solution.

bafonins, 2019-05-08:
* `FrontendError()` - can you provide examples of such errors?
* `BackendError()`

  > is instantiated with the error object returned from the backend

so which component instantiates the error? this might include creating additional module wrapper for the api in the console/oauth

Overall, i dont think we have to build all that error hierarchy and start small.
1. Handle fetching entity errors within the entity sections, e.g. failure when fetching an application when the user is on `/console/application/<application_id>/...`. This should be fairly simple and we have some groundwork for that ([what you mentioned](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/console/views/application/index.js#L140)). In such case we should render the error component instead of the entity pages.
1.1 We have properly formatted error -> show translated message
1.2 Something wen horribly wrong and we dont really know what happened -> show generic error.
2. Form submission can also result in errors, but i think we already cover that.
3. Handle errors within standalone components, like `<Events />` or `<GatewayStatistics />`. These components are container components and handle communication with the backend/other components on their own. So the easiest solution would be to come with the error styling and to override `componentDidCatch`. 
4. Errors embedded into html -> simply render the error page (https://github.com/TheThingsNetwork/lorawan-stack/issues/233)

These cases cover most of the problems we can have.

kschiffer, 2019-05-08:
> `FrontendError()` - can you provide examples of such errors?

It is true that more or less all anticipatable errors are actually originating from the backend. What I could envision as a case for `FrontendError()` is some high level try/catch to attempt to catch unhandled exceptions originating from frontend code. As in: if we have properly worked to catch all possible backend errors properly, the only thing left to occur is errors originating from frontend code. Examining the exception could then help us narrow down the cause and rethrowing proper `FrontendErrors()`.

> so which component instantiates the error? this might include creating additional module wrapper for the api in the console/oauth

The react components that catched/intercepted the error, eg.:
```js
// typical store error check
if (error) {
  // We know error is a backend error object
  throw new BackendError(error)
}
// or catch in a direct api call (like form submits)
…
catch (error) {
  throw new BackendError(error)
}
```
What is important is that we 
1. have an error class that provides us with a clean API to extract info from the error
2. use these errors to throw them up the react hierarchy / or to handle them in place
3. have streamlined error instances that we can pass to components as props so they will have everything needed to display info properly

> Overall, i dont think we have to build all that error hierarchy and start small.

I agree we can reduce the scaffolding by using a generic `ConsoleError()` rather that we can either pass either the backend error object or define our own errors in. The error class can then take care of exposing info we need in a clear API, by marshaling what it was instantiated with.
We can use this strategy for all of the cases you mentioned:

1. instanciate error and throw
2. instantiate error, pass to `Notifiaction` component (after updating the component logic, obviously)
3. instanciate error and throw
4. instantiate error in `PAGE_DATA` and throw

The exact display logic is then determined by the hierarchy of our error boundaries (which is a little out of scope for this)

What do you think?

bafonins, 2019-05-08:
> 
```js
// typical store error check
if (error) {
  // We know error is a backend error object
  throw new BackendError(error)
}
```

Why would you want to throw from the render method of an entity page? Isnt it easier to simply return the error component? We can even go further and add, say, `withError` hoc, that will check if there is an error, unwrap it and return the error component with descriptive/generic error message.

I think building a couple of react component that know how to handle specific object types, in this case errors, is easier that building error object hierarchy (backend <- console -> frontend). In most of the cases we know where the error came from, so i dont really see the purpose of distinguishing them. What we really care about is whether we can translate the message or not.

Let me know if i missing something.

kschiffer, 2019-05-08:
> Why would you want to throw from the render method of an entity page?

Yeah, we would not do it in the render method, but rather in `componentDidUpdate()`.

> Isnt it easier to simply return the error component? 

We would then have to use portals to be able to show full page errors. Otherwise, we could only display within the boundaries of the dom that the component is rendered into. E.g. in case of 404 of an entity we would still have to show the sidebar, which is not what we want. And even with portals, it would still render unnecessarily, unless we add logic to prevent that.
If we throw an error, it would bubble up to the nearest error boundary and render an error component there. Note that we don't have to throw the error per se, but rather only if we think that the issue cannot be resolved within the component.

> I think building a couple of react component that know how to handle specific object types, in this case errors, is easier that building error object hierarchy (backend <- console -> frontend).

Forget about the hierarchy, we can do it with just one error object, see my other comment:
> I agree we can reduce the scaffolding by using a generic ConsoleError() rather that we can pass either the backend error object or define our own errors in. The error class can then take care of exposing info we need in a clear API, by marshaling what it was instantiated with.

The logic flow would be like this:

If the issue is resolvable from within the component: 
⟶ use component's render function to display error info
Example: Form submit errors

If the issue is not resolvable from within the component: 
⟶throw an error to be caught by the nearest error boundary
Example: 404s

bafonins, 2019-05-08:
> We would then have to use portals to be able to show full page errors. Otherwise, we could only display within the boundaries of the dom that the component is rendered into. E.g. in case of 404 of an entity we would still have to show the sidebar, which is not what we want.

This can be avoided by the `withError` hoc i explained before. But propagating the error should work as well. Lets focus on implementing the groundwork for error handling and then we can discuss it further.

kschiffer, 2019-05-15:
Closed via #667

closed 2019-05-15 by kschiffer