TheThingsNetwork/lorawan-stack #669: Add error code to ErrorDetails

johanstokking, 2019-05-10:
#### Summary

Add `code` to `ErrorDetails`

#### Why do we need this?

To inform clients what the code is of errors and their causes, without knowing the TTN errors by namespace and name.

#### What is already there? What do you see now?

`ErrorDetails` with everything but the code.

The reason that the code isn't there, is that it's part of the `google.rpc.Status`, that's only returned on RPC and HTTP calls.

#### What is missing? What do you want to see?

We don't have `google.rpc.Status` available everywhere (i.e. event error payload) and we don't have the error code of the causes.

#### How do you propose to implement this?

Add `code` to `ErrorDetails` and fill out the field.

#### Can you do this yourself and submit a Pull Request?

Can do and can review.

johanstokking, 2019-07-09:
@rvolosatovs consider making this part of #967 as well

rvolosatovs, 2019-07-10:
I opened a draft PR for this https://github.com/TheThingsNetwork/lorawan-stack/pull/976, but testing blew up.
I will not make this part of #967

PR #976 (2019-07-16): Add error code to ErrorDetails

#### Summary

Closes #669 

#### Changes

- Add `Code` to `ErrorDetails` as well as encoding and decoding logic
closed 2019-07-16 by rvolosatovs