TheThingsNetwork/lorawan-stack #672: Creating a user through the cli fails

KolijnWolfaardt, 2019-05-12:
#### Summary
Creating a new user via ttn-lw-cli fails with error:
```error:pkg/identityserver/store:user_not_found (user `testuser` not found)```

#### Steps to Reproduce

Running on Ubuntu 18.04, with ttn-lw-cli installed via snap apps:

1. Deploy the stack as per the getting started guide, using  self-signed certificates, up to the 'Creating a gateway' section
2. Run ttn-lw-cli --ca cert.pem users create testuser
3. Fill in passwords
4. Error

#### What do you see now?
The full error reported by ttn-lw-cli is:
```
error:pkg/identityserver/store:user_not_found (user `testuser` not found)
    user_id=testuser
    correlation_id=0c161592b59c49bba7982512b76cab08
```
The Docker output shows nothing when the command is called. Other commands such as adding an application still work.

#### What do you want to see instead?
User created?

#### Environment

The Things Network Command-line Interface: ttn-lw-cli
Version:             3.0.3
Go version:          go1.12.5
OS/Arch:             linux/amd64

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

No

htdvisser, 2019-05-13:
This issue is caused by @johanstokking's https://github.com/TheThingsNetwork/lorawan-stack/pull/387. 

If you try to create a new user while also passing admin credentials, the Identity Server will do a lookup of the entity and return a NotFound if it doesn't exist. This is desired behavior for most calls, but obviously not for a create.

As a workaround, you can logout from the CLI before creating the new user.

johanstokking, 2019-06-20:
@adriansmares if rights validation errors with "not found" on create, don't treat that as an error.

adriansmares, 2019-07-01:
This is not as trivial as it initially seemed. The issue is that the rights check which fails is not from the actual gRPC registry call, which does not do any rights checks in the case of user create ([see here](https://github.com/TheThingsNetwork/lorawan-stack/blob/861fe5eaec412b8da0ff47cdb431b41e49a30b50/pkg/identityserver/user_registry.go#L89)), but rather from the rights fetcher hook, which is applied to all the calls ([see here](https://github.com/TheThingsNetwork/lorawan-stack/blob/861fe5eaec412b8da0ff47cdb431b41e49a30b50/pkg/identityserver/identityserver.go#L153-L172)).

One solution for this is to add the rights fetcher hook only to the other User Registry methods, and manually add the other two to the Create method (is this possible ?).

htdvisser, 2019-07-22:
This one got fixed as a side-effect of https://github.com/TheThingsNetwork/lorawan-stack/pull/1015

closed 2019-07-22 by htdvisser