TheThingsNetwork/lorawan-stack #686: Manage application webhooks in Console

johanstokking, 2019-05-14:
#### Summary

Manage application webhooks in the Console.

#### Why do we need this?

Webhooks are a powerful way to work with application data. It's on either side more scalable than MQTT. Also, webhooks allow for serverless applications.

#### What is already there? What do you see now?

AS supports webhooks natively. There's also management through API and CLI.

#### What is missing? What do you want to see?

Console support.

#### How do you propose to implement this?

Per application, be able to manage webhooks. This is CRUD, see: https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md#ttn.lorawan.v3.ApplicationWebhookRegistry

Per webhook, be able to manage the webhook (see https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md#ttn.lorawan.v3.ApplicationWebhook). In a presentable manner that is:
- Webhook ID, a user defined ID that uses the same rules for all our other IDs
- Format, get options through the `get-formats` endpoint, see API
- The base URL, to which paths are appended for specific message types. These message types are optional. I think we should have some kind of checkboxes per message type. When enabled, the path is optional too (can be empty), for example:
   - [ ] Uplink messages: ` ` (= disabled, so `uplink-message: null`)
   - [x] Join-accept messages: ` ` (= enabled, so `join-accept: { path: "" }`)
   - [x] Downlink queued: `/down/queued` (= enabled, so `downlink-queued: { path: "/down/queued" }`)
   - [ ] Downlink sent: see above
   - [ ] Downlink failed: see above
   - [ ] Downlink acknowledged: see above
   - [ ] Downlink not acknowledged: see above
   - [ ] Location solved: see above

#### Can you do this yourself and submit a Pull Request?

Can review.

kschiffer, 2019-05-16:
Is this console MVP?

johanstokking, 2019-05-16:
It would be _very_ nice to have.

kschiffer, 2019-05-22:
So, will these be treated as an integration, or should it be a dedicated section?

johanstokking, 2019-05-22:
Can you elaborate a bit?

kschiffer, 2019-05-22:
In the console, under single applications, we have an `Integrations` section. Should the webhooks functionality be made available under this section. Or should webhooks be treated in their own sub section.
[image]

So basically the question is: Are webhooks an »integration«?

johanstokking, 2019-05-22:
Ah, no that would be integrations

PR #771 (2019-06-06): Application Integrations (Webhooks)
#### Summary
Closes #686.

[image]
[image]
[image]

#### Changes
- Extend SDK services with webhooks
- Add toggled input component
- Add redux primitives for webhooks and webhook formats
- Add webhook-format selector container component
- Add webhook form component
- Add integrations/webhook views

#### Notes for Reviewers
- The redux logic here is a little naive but I didn't want to put a lot of thought into as we will refactor soon anyway

#### Release Notes
- Added views to handle integrations (webhooks) to the console

cc @johanstokking
closed 2019-06-06 by kschiffer