TheThingsNetwork/lorawan-stack #709: Dwell time requirements should be sent to devices on join

ama9910, 2019-05-20:
#### Summary
Devices should be sent Dwell Time requirements at time of Join.
Presumably this would be done using MAC command TxParamSetupReq.

This should be done whether the band requires Dwell or not, since the network server cannot know whether the device has it enabled or disabled by default.

#### Steps to Reproduce
1. Register and configure any device for LoRaWAN 1.0.2, OTAA, Class A. I use bandplan AS_920_923, but any band plan will probably work to reproduce.
2. Monitor event names for your app: ttn-lw-cli events --application-id yourAppID | jq .name
3. Issue a join request from your device
4. Note that in the event list, there is no TxParamSetupReq command
5. Note that whether your device is or isn't configured with a Dwell time restriction, there is no change after the join.

#### What do you see now?
Devices that enable dwell by default remain unchanged after the join.
Devices that disable dwell by default remain unchanged after the join.

#### What do you want to see instead?
Devices change their dwell setting on join to be in line with the band plan's dwell requirement, as a result of the network server sending them a TxParamSetupReq command

#### How do you propose to implement this?
Additional MAC command issued at time of Join

#### Can you do this yourself and submit a Pull Request?
Unfortunately not

johanstokking, 2019-05-20:
It looks like we need to send `TxParamSetupReq` after MAC state reset (e.g. join) when the device is in a band that supports dwell time.

@ama9910 is this a frequency plan setting or a per-device setting?

@rvolosatovs if it's a frequency plan, we need derivates of `AS_920_923`, `AS_923_925` and `AU_915_928_FSB_*` setting the dwell time. That's available on a frequency plan level as well as per-channel (but we don't need that).
If it's per device, we'd need new bools in `mac_settings` whether dwell time for uplink and downlink is desired.

johanstokking, 2019-05-20:
For the record; I would highly prefer making this a frequency plan setting.

ama9910, 2019-05-20:
Per-frequency-plan seems to be more closely aligned to the LoRaWAN spec, which expects the NS to dictate whether dwell is required, rather than the device/app owner.
Even though it means creating derivative band-plans.

Andrew

PR #719 (2019-05-27): Ensure TxParamSetupReq is scheduled when necessary

#### Summary

Closes #709 
Closes #690 

#### Changes

- Send `TxParamSetupReq` when variable(either on or off) dwell time requirement exists
- Fix `TxParamSetupSupport` value in AU915 bands

#### Notes for Reviewers

@johanstokking what do you think about this approach? Essentially `nil` for dwell-time field means that the value is unknown or irrelevant.
When the dwell-time field value is non-nil in desired MAC parameters, it means that both dwell-time options are possible in the band and the value denotes the actual "desired" value(from fre [... cut]
closed 2019-05-27 by rvolosatovs