TheThingsNetwork/lorawan-stack #723: Introduce update actions for console entities

bafonins, 2019-05-21:
#### Summary

Entities are not updated after the successful edit form submission which leads to stale data being shown in the ui.

#### Steps to Reproduce

1. Go the Application General Settings page
2. Update the application
3. Go to the Application Overview page and back to the settings page
4. See no changes being reflected in the ui

#### What do you see now?

No changes after successful updates.

#### What do you want to see instead?

The latest data after edits.

#### Environment

browser (console)

#### How do you propose to implement this?

Introduce update actions for console entities

#### Can you do this yourself and submit a Pull Request?

yes

kschiffer, 2019-05-22:
Would we connect this action also to the middleware to perform the update with the API?

bafonins, 2019-05-22:
> Would we connect this action also to the middleware to perform the update with the API?

Would be nice, however this can be challenging when dealing with forms.

kschiffer, 2019-05-22:
Yeah. I'm also wondering whether it makes sense to differentiate between patching and updating, or whether we include patching also for the update action. See also https://github.com/TheThingsNetwork/lorawan-stack/pull/708#discussion_r286434743

kschiffer, 2019-06-26:
So we currently already tie the update action for devices to the respective API call. This works well in most cases. However, one problem I identified is that update actions that delete fields will not result in a proper update of the store. The reason is that the responses of such update actions have no indication for the reducer as to what field should be removed because it is simply missing. If we spread the result over the existing store data, the deleted field will still be there.

To solve this, I think we need a reference to the original request action, which contains the patch of the update request. Using this and the request result, we should be able to determine which fields need to be removed.

Question is whether there is a simpler solution, or whether we should deal with these cases manually, be doing the request and calling the success actions ourselves. What do you think?

closed 2019-06-27 by bafonins