TheThingsNetwork/lorawan-stack #730: Add proper branding to oauth app

kschiffer, 2019-05-21:
#### Summary
The oauth app is currently not properly branded. It needs to be branded dynamically based on the stack configuration.
Related: https://github.com/TheThingsIndustries/lorawan-stack/issues/1041

#### Why do we need this?
Basic functionality. Unbranded OAuth app leaves users unclear about the proper context.

#### What is already there? What do you see now?
Some OAuth app pages with no logos, brand titles or whatsoever.

#### What is missing? What do you want to see?
At least a configurable logo and title, together with header and/or footer.

#### Environment
OAuth provider in browser.

#### How do you propose to implement this?
Subject of discussion.

#### Can you do this yourself and submit a Pull Request?
Yes.

johanstokking, 2019-09-11:
@kschiffer please indicate what the discussion is and mention people to loop them in

kschiffer, 2019-10-04:
Closing in favor of #1422

closed 2019-10-04 by kschiffer