TheThingsNetwork/lorawan-stack #752: Console Add Device page scrolling issue

furtiman, 2019-05-24:
#### Summary

While trying to add a device to an application on small screen (13"), not all the settings are seen, as well as the button `Create Device` to finish adding the device,  but scrolling is not possible (does not work).

On a big screen (27"), all the settings and the button are seen without the need to scroll.

#### Steps to Reproduce

1. Create an application
2. "Add Device"
3. ...

#### What do you see now?

13" screen: 
[image]

27" screen:
[image]

#### What do you want to see instead?

Scrolling enabled on the page

#### Environment

Safari

bafonins, 2019-05-24:
The issue seems to happen only when using safari.

Chaning [L25 app.styl](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/console/views/app/app.styl#L25) to `overflow: overlay` fixes the issue. This was changed in [this commit](https://github.com/TheThingsNetwork/lorawan-stack/commit/1ae922785dd2a9ea14b649107d2e6758b6652bdf#diff-0c75926bf7ddd31eca43d047867cf78c). @kschiffer could you justify the change?

kschiffer, 2019-05-24:
`overlay` is not a valid property for `overflow`. If I remember correctly, this was changed due to multiple scrollbars being visible otherwise.

kschiffer, 2019-05-24:
So seems like this needs a little more fiddling to get it right in all browsers.

bafonins, 2019-05-24:
> overlay is not a valid property for overflow.

Sort of. It is not standardized and works only in webkit based browsers, in others it fallbacks to `auto` as far as i understand.

That was the issue:
[image]

I checked latest versions of firefox/chrome and safari with `overflow: auto` - it solves the scrolling issue and doesnt revoke the 'multiple scrollba' issue

kschiffer, 2019-05-24:
I believe I also tried `auto` there but it was leading to some other problem. Maybe open a PR with it and I will double check.

PR #764 (2019-05-28): Fix console page scrolling in safari

#### Summary

Closes  #752
closed 2019-05-28 by bafonins